oliexdev/openScale #775: Add possibility to create new entries or add new important ones

0verreach3r, 2021-09-18:
Hi, I have a small feature request. The application is very complete and well functioning, but I think it's missing this component.

Under the **Settings>Measurements** section, it's possible to see, select and change the measurements order. Some of them are even pretty specific or rare to be useful to a standard users like waist-hip / waist-height ratio, all the caliper options...

Some very simple options like shoulder width, or the possibility to track the right/left one of a specific body part, are missing.

I don't know if it would be more simple to make a little menu which allows the user to add all the options he wants, since I think it would imply the developer to re-program an entire part of the application and how it works. Or if it would be more simple to add the options I listed before.

lindapi17, 2021-09-21:
Are you asking for additional types of measurement to be added. I know my scale, Eros by INEVIFIT, has other measurements not imported into the table like body age, protein, etc. And there are some I would like to add manually, like right and left for arm and leg measurements, and belly which is between hips and waist.

oliexdev, 2021-10-10:
Yeah this request comes from time to time, the measurement list could be endless. Perfect would be to add a more generic measurement but that is not easy to implement sorry, see #558.

closed 2021-10-10 by oliexdev