oliexdev/openScale #784: Android 12: crash by setting up a Bluetooth scale

ravn83, 2021-11-01:
Hello, i usw the New Pixel 6 with Android 12.
By setting up the App it crqshes by choosing the hlue tooth scale with the following error:

Build version: 2.4.1-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-01 12:48:01 
Device: Google Pixel 6 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@4f31dedb: GattService registerScanner
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2425)
	at android.os.Parcel.createException(Parcel.java:2409)
	at android.os.Parcel.readException(Parcel.java:2392)
	at android.os.Parcel.readException(Parcel.java:2334)
	at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerScanner(IBluetoothGatt.java:1727)
	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegistration(BluetoothLeScanner.java:426)
	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:278)
	at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:154)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:413)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:508)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:143)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:104)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3039)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:607)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:306)
	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:745)
	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:672)
	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:776)
	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:147)
	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:120)
	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:297)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2202)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2106)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:2002)
	at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7843)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

I hope you will fix it soon, so i can use the App with Android 12.

Greetings ravn

2low2free, 2021-11-02:
I had the same problem.  You can solve it by enabling nearby devices permission.

oliexdev, 2021-11-05:
thanks the Bluetooth permission handling was changed in Android 12, see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

I have pushed some changed to hopefully solve this problem but unfortunately I don't own an Android 12 smartphone to test these changes.

Could you try out the latest [openScale dev build](https://github.com/oliexdev/openScale/releases/download/travis-dev-build/openScale-dev-build.apk) on Android 12?

Please report back if it works or not. Thanks in advance!

I will close this issue, please use issue #779 .

closed 2021-11-05 by oliexdev

2low2free, 2021-11-05:
Thanks for reaching out. I installed the Dev apk and it worked with my Renpho scale. It asked for the location permission, which I granted this time only. It then said I needed to turn on the permission manually. I was able to search for the scale, it connected and recorded my weight. 

If you need me to pull logs, just let me know where they're located and I can provide them.

oliexdev, 2021-11-05:
Thanks but I was expecting that on Android 12 they will ask for permission BLUETOOTH SCAN and BLUETOOTH CONNECT and not anymore for the location permission. Could you revoke the permission and try to search for the scale in the Bluetooth settings again?

2low2free, 2021-11-06:
I reinstalled the app to start from a clean state. When I revoke the locaton permission, it won't let me progress to search for the scale. A toast message appears and states the permission is not granted and continues to loop.

oliexdev, 2021-11-07:
thanks, when you revoke the location permission then openScale will not ask again (its more an Android function) and you will see only the toast message that it was not granted. But what I wonder if they ask for the location permission. I was assumed that in Android >= 12 this will not ask anymore because it is not needed anymore only Bluetooth scan and Bluetooth connect permissions.
I added some debug information in the permission handling in the latest dev package. Could you install it and try it again while you [enabled the debug log of openScale](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-%28FAQ%29#how-to-get-the-openscale-debug-log-file)?

2low2free, 2021-11-08:
Here is the debug log

2021-11-08 12:45:50.431 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.3.2-dev_5d697fe9_2020-11-05 (1604584993), SDK 31, Google Pixel 5
2021-11-08 12:45:50.448 Debug [2] AboutPreferences: Selected user id(1) name(Bill) birthday(Thu Nov 08 12:43:33 EST 2001) age(20) body height(103.00) scale unit(lb) gender(male) initial weight(104.33) goal weight(97.52) goal date(Sun May 08 12:43:33 EDT 2022) measure unt(cm) activity level(0) assisted weighing(false)
2021-11-08 12:57:49.756 Info [2] BluetoothCentral: scan started
2021-11-08 12:57:52.035 Info [2] BluetoothCentral: scan stopped
2021-11-08 12:57:54.496 Info [2] BluetoothCentral: scan started
2021-11-08 12:57:57.855 Debug [2] BluetoothSettingsFragment: Found supported device QN-Scale [A4:C1:38:1A:40:0A] (driver: QN Scale)
2021-11-08 12:57:59.453 Debug [2] BluetoothSettingsFragment: Found unsupported device Amazfit Neo [F8:66:F3:5E:48:2A]
2021-11-08 12:58:02.677 Debug [2] BluetoothSettingsFragment$BluetoothDeviceView: Saved Bluetooth device QN-Scale with address A4:C1:38:1A:40:0A
2021-11-08 12:58:02.683 Info [2] BluetoothCentral: scan stopped
2021-11-08 12:58:02.697 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-11-08 12:58:16.491 Debug [2] OpenScale: Trying to connect to bluetooth device [A4:C1:38:1A:40:0A] (QN-Scale)
2021-11-08 12:58:16.499 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2021-11-08 12:58:16.511 Info [2] BluetoothCentral: scan started
2021-11-08 12:58:16.513 Debug [2] BluetoothCommunication: Stop machine state
2021-11-08 12:58:17.401 Debug [2] BluetoothCommunication: Found peripheral 'QN-Scale'
2021-11-08 12:58:17.404 Info [2] BluetoothCentral: scan stopped
2021-11-08 12:58:18.408 Debug [2] BluetoothCommunication: Try to connect to BLE device A4:C1:38:1A:40:0A
2021-11-08 12:58:18.518 Info [2] BluetoothPeripheral: connect to 'QN-Scale' (A4:C1:38:1A:40:0A) using TRANSPORT_LE
2021-11-08 12:58:19.569 Info [771] BluetoothPeripheral: connected to 'QN-Scale' (BOND_NONE) in 1.0s
2021-11-08 12:58:19.581 Debug [2] BluetoothPeripheral: discovering services of 'QN-Scale' with delay of 0 ms
2021-11-08 12:58:20.237 Info [771] BluetoothPeripheral: discovered 4 services for 'QN-Scale'
2021-11-08 12:58:20.244 Debug [2] BluetoothCommunication: connected to 'QN-Scale'
2021-11-08 12:58:20.248 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2021-11-08 12:58:20.251 Debug [2] BluetoothCommunication: Resume machine state
2021-11-08 12:58:20.254 Debug [2] BluetoothCommunication: Step Nr 0
2021-11-08 12:58:20.270 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2021-11-08 12:58:20.272 Debug [2] BluetoothCommunication: Stop machine state
2021-11-08 12:58:20.290 Debug [2] MainActivity: Bluetooth connection successful established
2021-11-08 12:58:20.355 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2021-11-08 12:58:20.356 Debug [2] BluetoothCommunication: Resume machine state
2021-11-08 12:58:20.358 Debug [2] BluetoothCommunication: Step Nr 1
2021-11-08 12:58:20.360 Debug [2] BluetoothCommunication: Invoke set indication on 0xffe2
2021-11-08 12:58:20.361 Debug [2] BluetoothCommunication: Stop machine state
2021-11-08 12:58:20.444 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe2-0000-1000-8000-00805f9b34fb
2021-11-08 12:58:20.446 Debug [2] BluetoothCommunication: Resume machine state
2021-11-08 12:58:20.447 Debug [2] BluetoothCommunication: Step Nr 2
2021-11-08 12:58:20.450 Debug [2] BluetoothCommunication: Invoke write bytes [13 09 15 02 10 00 00 00 43] on 0xffe3
2021-11-08 12:58:20.452 Debug [2] BluetoothCommunication: Step Nr 3
2021-11-08 12:58:20.455 Debug [2] BluetoothCommunication: Invoke write bytes [02 6C DB 1B 29] on 0xffe4
2021-11-08 12:58:20.457 Debug [2] BluetoothCommunication: Step Nr 4
2021-11-08 12:58:20.458 Debug [2] BluetoothCommunication: Step Nr 5
2021-11-08 12:58:20.460 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2021-11-08 12:58:20.465 Debug [2] BluetoothPeripheral: writing <130915021000000043> to characteristic <0000ffe3-0000-1000-8000-00805f9b34fb>
2021-11-08 12:58:20.474 Debug [2] MainActivity: Bluetooth scale message: Please step barefoot on the scale
2021-11-08 12:58:20.535 Debug [2] BluetoothCommunication: SUCCESS: Writing <13 09 15 02 10 00 00 00 43> to <0000ffe3-0000-1000-8000-00805f9b34fb>
2021-11-08 12:58:20.537 Debug [2] BluetoothCommunication: Step Nr 5
2021-11-08 12:58:20.539 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2021-11-08 12:58:20.544 Debug [2] BluetoothPeripheral: writing <026cdb1b29> to characteristic <0000ffe4-0000-1000-8000-00805f9b34fb>
2021-11-08 12:58:20.582 Debug [2] BluetoothQNScale: 14 0B 20 00 00 01 00 00 00 00 40 
2021-11-08 12:58:20.623 Debug [2] BluetoothCommunication: SUCCESS: Writing <02 6C DB 1B 29> to <0000ffe4-0000-1000-8000-00805f9b34fb>
2021-11-08 12:58:20.625 Debug [2] BluetoothCommunication: Step Nr 5
2021-11-08 12:58:20.626 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2021-11-08 12:58:20.804 Debug [2] BluetoothQNScale: 10 0B 20 29 72 00 00 00 00 00 D6 
2021-11-08 12:58:20.984 Debug [2] BluetoothQNScale: 10 0B 20 29 72 00 00 00 00 00 D6 
2021-11-08 12:58:21.211 Debug [2] BluetoothQNScale: 10 0B 20 29 72 00 00 00 00 00 D6 
2021-11-08 12:58:21.436 Debug [2] BluetoothQNScale: 10 0B 20 29 72 00 00 00 00 00 D6 
[... cut ...]
2021-11-08 12:58:34.010 Debug [2] BluetoothCommunication: disconnected 'QN-Scale' with status 0
2021-11-08 12:59:29.314 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-11-08 12:59:29.322 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-11-08 12:59:29.331 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-11-08 12:59:29.341 Error [2] BluetoothCentral: cannot cancel connection to unknown peripheral A4:C1:38:1A:40:0A
2021-11-08 12:59:29.391 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2021-11-09:
sorry, gradle has changed something so it was not the correctly updated dev apk file which you used but it should be fixed now. Please try out the latest [openScale dev apk](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) again.

2low2free, 2021-11-10:
I get this when trying to install the new Dev apk.

oliexdev, 2021-11-10:
@2low2free can't see anything in your post? Did you attach something?

2low2free, 2021-11-10:
My apologies, I keep getting "There was a problem parsing the package" when I try to install the apk.

oliexdev, 2021-11-10:
Please try out the following
***@***.***/how-to-fix-problem-parsing-the-package-error-in-android-c2fca91673cb

oliexdev, 2021-11-11:
or https://www.androiddata-recovery.com/blog/fix-there-was-a-problem-parsing-the-package-error

Did you allowed the install from unknown sources?

2low2free, 2021-11-11:
Yes. It was already selected when I checked my settings. I've done 1,2,3 and 7. It still doesn't install. 

My phone has updated in the last week.

oliexdev, 2021-11-11:
maybe you can check the md5 checksum of the downloaded apk file.
It should be

```
> md5sum openScale-dev-build.apk 
48a915888afb59a86a71449459c15e73  openScale-dev-build.apk
```

2low2free, 2021-11-11:
Hashes match. Here are screenshots.

oliexdev, 2021-11-11:
then I have no idea anymore. I can't reproduce this error here but I don't have a real Android 12 device here.

2low2free, 2021-11-11:
I'll download it on my PC and then transfer it to my phone to rule out an incomplete download

oliexdev, 2021-11-12:
found the problem, the export feature in the Android 12 Manifest is now required. In the released 2.4.3 this problem should be fixed.

2low2free, 2021-11-12:
It only asked for location of nearby devices

2021-11-12 15:24:36.298 Debug [2] AboutPreferences: Debug log enabled, openScale (dev) v2.4.3-dev_49ae42b5_2021-11-12 (1636745776), SDK 31, Google Pixel 5
2021-11-12 15:24:36.319 Debug [2] AboutPreferences: Selected user id(1) name(Bill) birthday(Mon Nov 12 15:23:32 EST 2001) age(20) body height(108.00) scale unit(lb) gender(male) initial weight(104.33) goal weight(97.52) goal date(Thu May 12 15:23:32 EDT 2022) measure unt(cm) activity level(0) assisted weighing(false)
2021-11-12 15:24:44.717 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2021-11-12 15:24:48.038 Debug [2] BluetoothSettingsFragment: At least one Bluetooth permission was not granted
2021-11-12 15:24:50.863 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2021-11-12 15:25:06.216 Debug [2] BluetoothSettingsFragment: All Bluetooth permissions granted
2021-11-12 15:25:06.463 Info [2] BluetoothCentralManager: scan started
2021-11-12 15:25:06.475 Debug [2] PermissionHelper: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2021-11-12 15:25:06.514 Debug [2] BluetoothSettingsFragment: All Bluetooth permissions granted
2021-11-12 15:25:06.522 Info [2] BluetoothCentralManager: scan started
2021-11-12 15:25:09.411 Debug [2] BluetoothSettingsFragment: Found supported device QN-Scale [A4:C1:38:1A:40:0A] (driver: QN Scale)
2021-11-12 15:25:11.115 Debug [2] BluetoothSettingsFragment: Found unsupported device Amazfit Neo [F8:66:F3:5E:48:2A]
2021-11-12 15:25:14.865 Debug [2] BluetoothSettingsFragment$BluetoothDeviceView: Saved Bluetooth device QN-Scale with address A4:C1:38:1A:40:0A
2021-11-12 15:25:14.878 Info [2] BluetoothCentralManager: scan stopped
2021-11-12 15:25:14.890 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2021-11-12 15:25:26.471 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2021-11-12 15:25:26.475 Error [2] BluetoothSettingsFragment: Fragment BluetoothSettingsFragment{3438ed6} (2c2af55a-c8ef-4b6c-8de1-9918721a4826) not attached to a context.