oliexdev/openScale #785: "weeks" are 10 days, not 7

TinfoilSubmarine, 2021-11-01:
**Describe the bug**
Chart shows "weeks" in groups of 10 days, not 7.

**To Reproduce**
Steps to reproduce the behavior:
1. Select "filter" button on top right in chart view.
2. Click on "Week view".
3. Compare two days on week dividers (e.g. W39 2021 and W40 2021)
4. See that according to chart, "W39 2021" starts on "Sep 20, 2021" while "W40 2021" starts on "Sep 30, 2021".

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???

**Expected behavior**
If "W39 2021" starts on "Sep 20, 2021", then "W40 2021" should start on "Sep 27, 2021". I would expect that these week numbers would correspond to ISO week standard, which would mean that W39 should start on Sep 27, 2021 and W40 should start on Oct 4, 2021.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

oliexdev, 2021-11-05:
that's really complex, it is not only depending with the Date format in Android also it depends with the MPAndroid library viewport handling. You can look into the following [source code ](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/gui/measurement/ChartMeasurementView.java) and try to improve it.

closed 2023-07-22 by oliexdev

TinfoilSubmarine, 2023-08-07:
I'm still having this issue with 2.5.2 installed.

TinfoilSubmarine, 2023-08-08:
So I've started to do some diagnosis and research and what I've found is that week view, month view, and year view are all skewed/incorrect.

One instance of problems with the year view has been specifically noted in https://github.com/oliexdev/openScale/issues/535. I think this is all due to how the underlying library formats, since there is no control over what values are used as labels, meaning that May 2023 could be shown on any date between May 1st and May 31st. The same principle applies to years and week numbers.

One potential solution I'm looking into is [using LimitLine](https://weeklycoding.com/mpandroidchart-documentation/axis-general/) which allows to specify a specific value. We could then generate the set of values needed (W1 2023 = 2023-01-02 ... W41 2023 = 2023-10-09) and create that many limit lines.