oliexdev/openScale #786: Support for Withings WS50

ameisehaufen, 2021-11-04:
Could we have support for the Withings WS-50 scale? How can I help to provide info to make it possible?

oliexdev, 2021-11-05:
please read https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

Note that the Withings WS-50 works with WLAN or with Bluetooth but openScale supports only Bluetooth scales.

ameisehaufen, 2021-11-05:
Thanks for the reply, I do some research in WS50 and found that it could work only in bluetooth, WIFI is optional, so, I will try to provide the information needed. I need some time to do that, I hope the bluetooth function in WS50 are compatible with OpenScale.
According to TOM's guide, wiithings body plus are the best smart scale in the market, and it is probably the most sold one, if it were supported, it would bring more users to OpenScale.

yabdali, 2022-06-25:
> Thanks for the reply, I do some research in WS50 and found that it could work only in bluetooth, WIFI is optional, so, I will try to provide the information needed. I need some time to do that, I [...]

Hi, has there been any progress on this? I have WS50 for few years and been waiting for it to be included.  Appreciate your efforts, thanks...

ameisehaufen, 2022-09-05:
> Hi, has there been any progress on this? I have WS50 for few years and been waiting for it to be included. Appreciate your efforts, thanks...

Hi, I'm just waiting for my next vacation to work on this... until then I'm downloading the data from wiithings manually.

Susurrus, 2023-07-09:
I just got me one of these and have been trying to follow [the instructions](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale) for how to enable support (thanks so much for writing some up!). I ran into problems with both Steps 1 & 2 on my Samsung S9 running Android 10.

For Step 1 I couldn't get a BT log from the phone. I may have another one that would work, so I'll try that later if no one else beats me to it.

For Step 2, the dev app didn't give me any output that was useful:
```
2023-07-09 17:50:05.380 Debug [2] AboutPreferences: Debug log enabled, openScale v2.5.1 (63), SDK 29, samsung SM-G960U
2023-07-09 17:50:05.391 Debug [2] AboutPreferences: Selected user id(0) name() birthday(Sun Jul 09 17:50:05 GMT+02:00 2023) age(0) body height(-1.00) scale unit(kg) gender(male) initial weight(-1.00) goal enabled(false) goal weight(-1.00) goal date(Sun Jul 09 17:50:05 GMT+02:00 2023) measure unt(cm) activity level(0) assisted weighing(false)
2023-07-09 17:50:17.437 Debug [2] BluetoothSettingsFragment: Bluetooth settings Bluetooth permission check
2023-07-09 17:50:17.441 Debug [2] BluetoothSettingsFragment: SDK >= 29 request for Access fine location permission
2023-07-09 17:50:17.442 Debug [2] BluetoothSettingsFragment: Permission already granted: android.permission.ACCESS_FINE_LOCATION
2023-07-09 17:50:17.449 Info [2] BluetoothCentralManager: scan started
2023-07-09 17:50:36.327 Info [2] BluetoothCentralManager: scan stopped
2023-07-09 17:50:56.762 Debug [2] BluetoothSettingsFragment: Bluetooth settings Bluetooth permission check
2023-07-09 17:50:56.765 Debug [2] BluetoothSettingsFragment: SDK >= 29 request for Access fine location permission
2023-07-09 17:50:56.767 Debug [2] BluetoothSettingsFragment: Permission already granted: android.permission.ACCESS_FINE_LOCATION
2023-07-09 17:50:56.777 Info [2] BluetoothCentralManager: scan started
2023-07-09 17:51:16.785 Info [2] BluetoothCentralManager: scan stopped
2023-07-09 17:51:21.428 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-07-09 17:51:22.360 Debug [2] BluetoothSettingsFragment: Bluetooth settings Bluetooth permission check
2023-07-09 17:51:22.363 Debug [2] BluetoothSettingsFragment: SDK >= 29 request for Access fine location permission
2023-07-09 17:51:22.364 Debug [2] BluetoothSettingsFragment: Permission already granted: android.permission.ACCESS_FINE_LOCATION
2023-07-09 17:51:22.370 Info [2] BluetoothCentralManager: scan started
2023-07-09 17:51:22.394 Debug [2] BluetoothSettingsFragment: Found unsupported device Withings WS50 A4 [00:24:E4:1E:D6:A5]
2023-07-09 17:51:23.753 Info [2] BluetoothCentralManager: scan stopped
2023-07-09 17:51:23.754 Debug [2] OpenScale: Trying to connect to bluetooth device [00:24:E4:1E:D6:A5] in debug mode
2023-07-09 17:51:23.758 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2023-07-09 17:51:23.764 Info [2] BluetoothCentralManager: scan started
2023-07-09 17:51:23.765 Debug [2] BluetoothCommunication: Stop machine state
2023-07-09 17:51:23.794 Debug [2] BluetoothCommunication: Found peripheral 'Withings WS50 A4'
2023-07-09 17:51:23.797 Info [2] BluetoothCentralManager: scan stopped
2023-07-09 17:51:24.804 Debug [2] BluetoothCommunication: Try to connect to BLE device 00:24:E4:1E:D6:A5
2023-07-09 17:51:24.923 Info [2] BluetoothPeripheral: connect to 'Withings WS50 A4' (00:24:E4:1E:D6:A5) using transport LE
2023-07-09 17:51:24.962 Info [2] BluetoothPeripheral: peripheral '00:24:E4:1E:D6:A5' is connecting
2023-07-09 17:51:30.005 Info [8453] BluetoothPeripheral: connection failed with status 'CONNECTION_FAILED_ESTABLISHMENT'
2023-07-09 17:51:30.020 Info [8453] BluetoothCentralManager: connection to 'Withings WS50 A4' (00:24:E4:1E:D6:A5) failed
2023-07-09 17:51:30.026 Error [2] BluetoothCommunication: connection 'Withings WS50 A4' failed with status 62
2023-07-09 17:51:30.028 Debug [2] OpenScale: Disconnecting from bluetooth device
2023-07-09 17:51:30.030 Debug [2] BluetoothCommunication: Bluetooth disconnect
2023-07-09 17:51:30.030 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-07-09 17:51:41.277 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-07-09 17:51:41.278 Debug [2] OpenScale: Trying to connect to bluetooth device [00:24:E4:1E:D6:A5] in debug mode
2023-07-09 17:51:41.280 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2023-07-09 17:51:41.286 Info [2] BluetoothCentralManager: scan started
2023-07-09 17:51:41.287 Debug [2] BluetoothCommunication: Stop machine state
2023-07-09 17:51:41.433 Debug [2] BluetoothCommunication: Found peripheral 'Withings WS50 A4'
2023-07-09 17:51:41.439 Info [2] BluetoothCentralManager: scan stopped
2023-07-09 17:51:42.444 Debug [2] BluetoothCommunication: Try to connect to BLE device 00:24:E4:1E:D6:A5
2023-07-09 17:51:42.562 Info [2] BluetoothPeripheral: connect to 'Withings WS50 A4' (00:24:E4:1E:D6:A5) using transport LE
2023-07-09 17:51:42.582 Info [2] BluetoothPeripheral: peripheral '00:24:E4:1E:D6:A5' is connecting
2023-07-09 17:51:47.612 Info [8453] BluetoothPeripheral: connection failed with status 'CONNECTION_FAILED_ESTABLISHMENT'
2023-07-09 17:51:47.630 Info [8453] BluetoothCentralManager: connection to 'Withings WS50 A4' (00:24:E4:1E:D6:A5) failed
2023-07-09 17:51:47.636 Error [2] BluetoothCommunication: connection 'Withings WS50 A4' failed with status 62
[... cut ...]
2023-07-09 17:51:47.639 Debug [2] BluetoothCommunication: Bluetooth disconnect
2023-07-09 17:51:47.640 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-07-09 17:52:00.514 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2023-07-09 17:52:06.897 Debug [2] AboutPreferences: Debug log disabled
```

The BLE Scanner app also wasn't able to connect to it. Has anyone else tried this to see if anything works here? I wonder if there's some crude authentication that Withings does here...

Susurrus, 2023-07-09:
Okay, managed to get the snooper working. For the S9 it's in `FS/data/log/bt`.

I've been trying to find any of my measurements Wireshark and it's not working too well. I have the numbers the scale and app reported (imperial and metric), but haven't found anything. Would anyone be able to help me investigate this?

Susurrus, 2023-07-10:
Something that happens at the beginning of connecting to the scale is that the scale dumps a bunch of user data. Packets look like:
```
0000   02 0b 20 57 00 53 00 40 00 29 ff 9d 01 01 01 00   .. W.S.@.)......
0010   4a 01 01 0d 00 45 01 12 00 41 40 7b 22 73 74 61   J....E...A@{"sta
0020   74 75 73 22 3a 30 2c 22 62 6f 64 79 22 3a 7b 22   tus":0,"body":{"
0030   73 65 73 73 69 6f 6e 69 64 22 3a 22 37 50 78 4f   sessionid":"7PxO
0040   46 46 35 67 33 79 6c 79 53 2d 67 33 41 65 42 62   FF5g3ylyS-g3AeBb
0050   67 77 2d 59 5a 45 6c 66 66 54 68 76               gw-YZElffThv
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 22 2c 22 73 70 22   ....E...A@","sp"
0020   3a 7b 22 75 73 65 72 73 22 3a 5b 7b 22 69 64 22   :{"users":[{"id"
0030   3a 36 39 33 33 30 34 32 2c 22 73 6e 22 3a 22 53   :6933042,"sn":"S
0040   54 45 22 2c 22 77 74 22 3a 37 30 2e 38 34 39 2c   TE","wt":70.849,
0050   22 68 74 22 3a 31 2e 38 31 2c 6a                  "ht":1.81,j
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 22 61 67 74 22 3a   ....E...A@"agt":
0020   34 32 2e 31 2c 22 73 78 22 3a 30 2c 22 66 6d 22   42.1,"sx":0,"fm"
0030   3a 31 33 31 2c 22 63 72 22 3a 31 35 33 39 39 32   :131,"cr":153992
0040   34 31 31 37 2c 22 61 74 74 22 3a 30 2c 22 66 74   4117,"att":0,"ft
0050   22 3a 5b 5d 2c 22 66 6e 22 3a 6a                  ":[],"fn":j
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 22 53 74 65 66 61   ....E...A@"Stefa
0020   6e 22 2c 22 63 73 22 3a 22 31 31 2c 36 2c 31 2c   n","cs":"11,6,1,
0030   31 30 2c 33 2c 34 22 7d 2c 7b 22 69 64 22 3a 36   10,3,4"},{"id":6
0040   39 33 33 36 33 31 2c 22 73 6e 22 3a 22 46 52 41   933631,"sn":"FRA
0050   22 2c 22 77 74 22 3a 35 34 2e 6a                  ","wt":54.j
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 33 35 2c 22 68 74   ....E...A@35,"ht
0020   22 3a 31 2e 36 31 2c 22 61 67 74 22 3a 33 39 2e   ":1.61,"agt":39.
0030   39 2c 22 73 78 22 3a 31 2c 22 66 6d 22 3a 33 2c   9,"sx":1,"fm":3,
0040   22 63 72 22 3a 31 35 33 32 32 34 30 39 30 36 2c   "cr":1532240906,
0050   22 61 74 74 22 3a 30 2c 22 66 6a                  "att":0,"fj
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 74 22 3a 5b 5d 2c   ....E...A@t":[],
0020   22 66 6e 22 3a 22 46 72 61 6e 7a 69 73 6b 61 22   "fn":"Franziska"
0030   2c 22 63 73 22 3a 22 31 31 2c 36 2c 31 2c 31 30   ,"cs":"11,6,1,10
0040   2c 33 2c 34 22 7d 2c 7b 22 69 64 22 3a 33 35 32   ,3,4"},{"id":352
0050   37 39 36 36 32 2c 22 73 6e 22 6a                  79662,"sn"j
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 3a 22 4a 41 4e 22   ....E...A@:"JAN"
0020   2c 22 77 74 22 3a 37 36 2e 30 31 36 2c 22 68 74   ,"wt":76.016,"ht
0030   22 3a 31 2e 38 32 2c 22 61 67 74 22 3a 32 30 2c   ":1.82,"agt":20,
0040   22 73 78 22 3a 31 2c 22 66 6d 22 3a 33 2c 22 63   "sx":1,"fm":3,"c
0050   72 22 3a 31 36 38 38 39 30 39 6a                  r":1688909j
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 33 32 34 2c 22 61   ....E...A@324,"a
0020   74 74 22 3a 30 2c 22 66 74 22 3a 5b 5d 2c 22 66   tt":0,"ft":[],"f
0030   6e 22 3a 22 4a 61 6e 65 22 2c 22 63 73 22 3a 22   n":"Jane","cs":"
0040   31 31 2c 31 2c 33 22 7d 5d 7d 2c 22 69 6e 64 22   11,1,3"}]},"ind"
0050   3a 7b 22 6c 67 22 3a 22 65 6e 6a                  :{"lg":"enj
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 5f 55 53 22 2c 22   ....E...A@_US","
0020   69 6d 74 22 3a 31 2c 22 73 74 70 22 3a 30 2c 22   imt":1,"stp":0,"
0030   66 22 3a 30 2c 22 67 22 3a 39 38 30 38 33 2c 22   f":0,"g":98083,"
0040   74 6d 70 22 3a 31 31 2c 22 77 22 3a 31 2c 22 66   tmp":11,"w":1,"f
0050   74 22 3a 5b 5d 7d 2c 22 73 79 6a                  t":[]},"syj
```
```
0000   02 0b 20 56 00 52 00 40 00 29 ef 9d 01 01 00 4a   .. V.R.@.).....J
0010   01 01 0d 00 45 01 12 00 41 40 70 22 3a 7b 22 75   ....E...A@p":{"u
0020   74 63 22 3a 31 36 38 38 39 31 39 35 30 32 7d 2c   tc":1688919502},
0030   22 63 74 70 22 3a 7b 22 67 6f 66 66 22 3a 37 32   "ctp":{"goff":72
0040   30 30 2c 22 64 73 74 22 3a 31 36 39 38 35 34 31   00,"dst":1698541
0050   32 30 30 2c 22 6e 67 6f 66 66 6a                  200,"ngoffj
```
```
0000   02 0b 20 31 00 2d 00 40 00 29 ef 53 01 01 00 25   .. 1.-.@.).S...%
0010   01 01 0d 00 20 01 12 00 1c 1b 22 3a 33 36 30 30   .... .....":3600
0020   7d 2c 22 63 6f 6c 6f 72 22 3a 22 62 6c 61 63 6b   },"color":"black
0030   22 7d 7d 5c 30 6a                                 "}}\0j
```

I see a bunch of binary data packets after that, so I'm assuming that's the raw data.

Susurrus, 2023-07-10:
JSON is nicely null-terminated, so that's convenient. Collecting all the JSON and pretty-printing it:
```

{
  "status": 0,
  "body": {
    "sessionid": "7PxOFF5g3ylyS-g3AeBbgw-YZElffThv",
    "sp": {
      "users": [
        {
          "id": 6933042,
          "sn": "STE",
          "wt": 70.849,
          "ht": 1.81,
          "agt": 42.1,
          "sx": 0,
          "fm": 131,
          "cr": 1539924117,
          "att": 0,
          "ft": [],
          "fn": "Stefan",
          "cs": "11,6,1,10,3,4"
        },
        {
          "id": 6933631,
          "sn": "FRA",
          "wt": 54.35,
          "ht": 1.61,
          "agt": 39.9,
          "sx": 1,
          "fm": 3,
          "cr": 1532240906,
          "att": 0,
          "ft": [],
          "fn": "Franziska",
          "cs": "11,6,1,10,3,4"
        },
        {
          "id": 35279662,
          "sn": "JAN",
          "wt": 76.016,
          "ht": 1.82,
          "agt": 20,
          "sx": 1,
          "fm": 3,
          "cr": 1688909324,
          "att": 0,
          "ft": [],
          "fn": "Jane",
          "cs": "11,1,3"
        }
      ]
    },
    "ind": {
      "lg": "en_US",
      "imt": 1,
      "stp": 0,
      "f": 0,
      "g": 98083,
      "tmp": 11,
      "w": 1,
      "ft": []
    },
    "syp": {
      "utc": 1688919502
    },
    "ctp": {
      "goff": 7200,
      "dst": 1698541200,
      "ngoff": 3600
    },
    "color": "black"
  }
}
```

Susurrus, 2023-07-10:
Decided to email Withings in parallel and see if they'll tell us the protocol. Worth a shot!

mannp, 2023-11-19:
I have one of these scales and hoping it might get support 🫰🏻

iMonZ, 2024-12-16:
That would be awesome! And Bluetooth only would be fine

Susurrus, 2024-12-17:
Withings has said they can't provide details on their protocol. Something about it being proprietary blah blah blah. Unfortunately I don't have the email anymore to provide the exact wording. I have no plans to reverse engineer this anymore, as I'm not doing any home automation nor really need this tracking (I just manually input it into my tracking app). So if someone wants this to happen, I suggest creating a bounty to get someone to do it otherwise you'll need to do the legwork here. I'm happy to help test should someone make something though.

Susurrus, 2024-12-17:
Okay, found the communication, it wasn't an email it was in their internal support comms system on their website:
[image]