oliexdev/openScale #787: Xiaomi Mi Scale 2 - showing lean body mass instead of muscle mass

kauzu, 2021-11-05:
**Describe the bug**
The Xiaomi Mi Scale 2 sends lean body mass instead of muscle mass.
I'm not sure if this is a bug in the scale, or in the apps (both the Mi Fit app + OpenScale)

**To Reproduce**
Steps to reproduce the behavior:
 No specific Steps. This is always the case with MiScale2 

Reproduced with latest F-Droid app + all previous versions i used

**Expected behavior**
I expect to see my muscle mass

**Additional context**
I'm using OpenScale + MiScale2 for about 10 Month now. From the beginning it was clear that there might be something wrong with the shown muscle mass/percentage, but i didn't really care as this is a "cheap" scale and muscle mass was a bit irrelevant for me. 
In the meantime my body changed and the muscle mass shown in the app had gotten really unrealistic. 
I googled a bit and found this thread. https://www.reddit.com/r/Xiaomi/comments/ilfnlp/mi_scale_2_mistaking_muscle_mass_for_lean_mass/  
i did some math and can confirm that in openscale (and MiFit) the musclemass == Weight - body fat - bonemass
so what is shown as musclemass is actually the lean body mass. 
[image]

Hopefully there is some other parameter in the protocol which is the actual musclemass. if not, it would at least be fine if openscale would show the lean body mass as lean body mass and leave the musclemass empty (or do some sane assumtions based on the other data)

**Debug log**
if you need some bluetooth sniffing or so let me know.

oliexdev, 2021-11-05:
referring to https://www.healthline.com/health/muscle-mass-percentage

>  People often use the terms “lean body mass” and “muscle mass” interchangeably, but they’re not the same. Lean body mass includes muscle mass, as well as bones and bodily fluid. 

Anyway even if we assign it to lean body mass, I promise somebody will wonder why the values are differently between the vendors app and openScale.

Focshole, 2022-02-17:
What about instead changing the reference parameters so that they won't look as always "too much" even when they are not?
[image]

ruihildt, 2022-07-08:
> Anyway even if we assign it to lean body mass, I promise somebody will wonder why the values are differently between the vendors app and openScale.

In this case, I think that would not be a bad thing, because the measurement reported is completely bonker: when users research Muscle Mass, they'll realize the mesure doesn't add up.

If it's not possible to have the label adjusted for this scale, it would then be better to not populate thus "wrong" data.

ruihildt, 2022-07-08:
I looked through the closed issues, and several users of this scale have been similarly baffled by this labeling error (even though they didn't know at the time what the error was).

@oliexdev Would you accept a PR fixing this label issue?

Frans-Willem, 2022-07-21:
Also affected by this issue, and would like to see it resolved. Would love a build that has this fixed, even if I'd have to build it myself.
The fact that MiFit reports wrong data doesn't mean OpenScale should too :/

Focshole, 2022-12-19:
I have proposed a fix, rather than touching the muscle parameter, we can set the lean mass parameter too. I had found some reference parameters on pubmed. Feel free to review it and thank you @oliexdev for your amazing work so far!

PR #906 (2023-01-20): Miscale2 lean mass fix
Fixes #787 .

Rather than changing the muscle value, we can fill the lean mass value which was already implemented, in such a way we do not contradict what is being shown in the mifit app.
This PR adds some reference values for lean mass from a [pubmed paper]( https://pubmed.ncbi.nlm.nih.gov/30536335/) too.
closed 2023-01-20 by oliexdev