oliexdev/openScale #788: BMI index clarification

alejandracios, 2021-11-05:
Hi, 
I have been using the app and found that BMI has no references on the index. It is possible to add how it is calculated and what is the reference of that index? I have looking on internet and the scale in the app is not fully compatible with all the BMI calculations arround there

Thanks.

oliexdev, 2021-11-05:
the formula in openScale is BMI = weight / height^2 see https://en.wikipedia.org/wiki/Body_mass_index

closed 2021-11-05 by oliexdev

alejandracios, 2021-11-05:
What I meant is on the categories are being used. Wikipedia shows a categories going from: 
Category	BMI (kg/m2)[c]	
Underweight (Severe thinness)	< 16.0	
Underweight (Moderate thinness)	16.0 – 16.9	
Underweight (Mild thinness)	17.0 – 18.4	
Normal range	18.5 – 24.9	
Overweight (Pre-obese)	25.0 – 29.9	
Obese (Class I)	30.0 – 34.9	
Obese (Class II)	35.0 – 39.9	
Obese (Class III)	≥ 40.0	

While the app only a scale going from 17.0 ------ 22.0--------27.0-------32
which does not 100% match the BMI categories. This is why would like to have som sort of clarification. 

> the formula in openScale is BMI = weight / height^2 see https://en.wikipedia.org/wiki/Body_mass_index

hackel, 2022-01-02:
I agree @oliexdev, I have not seen these numbers used before an and was curious where they came from.