oliexdev/openScale #789: Beurer BF720

schweigel, 2021-11-07:
Beurer BF720

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex: male
body height: 182cm
age: 37
activity level: 3
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
weight: 94,6 kg
date: 07.11.2021 14:04
water: 48,3 %
muscle: 37,7 %
fat: 26,8 %
bones: 3,5 kg
basal metabolic rate: 1903 kcal
activity over the basal metabolic rate: 2855 kcal
```
[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/7492344/btsnoop_hci.log)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex: male
body height: 182cm
age: 37
activity level: 3
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
weight: 97,0 kg
date: 07.11.2021 14:12
water: 47,3 %
muscle: 37,1 %
fat: 28,0 %
bones: 3,6 kg
basal metabolic rate: 1927 kcal
activity over the basal metabolic rate: 2698 kcal
```

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/7492356/btsnoop_hci.log)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex: male
body height: 182cm
age: 37
activity level: 3
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
weight: 95,8 kg
date: 07.11.2021 14:17
water: 47,7 %
muscle: 37,4 %
fat: 27,5 %
bones: 3,6 kg
basal metabolic rate: 1915 kcal
activity over the basal metabolic rate: 2873 kcal
```

[btsnoop_hci.log](https://github.com/oliexdev/openScale/files/7492357/btsnoop_hci.log)

**Step 3: Discover Bluetooth services and characteristic**

[openScale_2021-11-04_20-09.txt](https://github.com/oliexdev/openScale/files/7492358/openScale_2021-11-04_20-09.txt)

oliexdev, 2021-11-08:
please try out the latest [dev build version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build). Maybe we have luck and it works like the BF105 scale.

schweigel, 2021-11-08:
Hi,

I want to try the dev build but it says that BF720 is not supported. I'm not sure but info of APP says the Version is 2.3.2-dev_5d697fe9_2020-11-05. In the download page is a different commit mentioned. Maybe I have the wrong dev Version?

oliexdev, 2021-11-09:
thanks for the hint gradle has changed something so it wouldn't correctly update the  dev apk file but it should be fixed now. Please try out the latest [openScale dev apk](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build) again.

schweigel, 2021-11-09:
Hi,

it seams we have some luck 😃 

1. connecting the scale works, but the scale doesn't show the pin like in beurer app. So it is nessesary to use buttons on scale to show the right pin for the connection
2. data transfer for weight, water, muscle, fat, bones, BMI works (scale shows only one decimal places and openscale shows two, so small differences(<0,1) in the decimal places are fine)
3. basal metabolic rate and activity over the basal metabolic rate are different on scale and openscale

Maybe it is possible to solve the open issues, but for the beginning it is great.

kind regards

oliexdev, 2021-11-10:
> Hi,
> 
> it seams we have some luck smiley
> 
>     1. connecting the scale works, but the scale doesn't show the pin like in beurer app. So it is nessesary to use buttons on scale to show the right pin for the connection
> 

but that should be no problem.

>     2. data transfer for weight, water, muscle, fat, bones, BMI works (scale shows only one decimal places and openscale shows two, so small differences(<0,1) in the decimal places are fine)
> 

that's really good, I guess the Beurer app just round the value, no action needed.

>     3. basal metabolic rate and activity over the basal metabolic rate are different on scale and openscale
> 

openScale use for *all* scales the Harris-Benedict equation 

> 
> Maybe it is possible to solve the open issues, but for the beginning it is great.
> 
> kind regards

closed 2021-11-10 by oliexdev

schweigel, 2021-11-11:
> > Hi,
> > it seams we have some luck smiley
> > ```
> > 1. connecting the scale works, but the scale doesn't show the pin like in beurer app. So it is nessesary to use buttons on scale to show the right pin for the connection
> > ```
> 
> but that should be no problem.

Yes, but it would be nice to have this feature. I think it should be only send a request for that to the scale. Maybe adding this as a new feature request?

> 
> > ```
> > 2. data transfer for weight, water, muscle, fat, bones, BMI works (scale shows only one decimal places and openscale shows two, so small differences(<0,1) in the decimal places are fine)
> > ```
> 
> that's really good, I guess the Beurer app just round the value, no action needed.
> 

👍😁

> > ```
> > 3. basal metabolic rate and activity over the basal metabolic rate are different on scale and openscale
> > ```
> 
> openScale use for _all_ scales the Harris-Benedict equation
> 

I don't know anything about the calculations for that, but I think it would be nice to have the choice for the calculation in app or to use the values from the scale like switch bones in kg or %.

> > Maybe it is possible to solve the open issues, but for the beginning it is great.
> > kind regards

Finally, thank you for your great work.