oliexdev/openScale #791: Backup crashing on Android 12

rodakswe, 2021-11-14:
**Describe the bug**
App crashes when backup is clicked.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on 'Backup'
3. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): 2.4.3-dev_49ae42b5_2021-11-12

**Expected behavior**
Backup menu should open.

**Additional context**

**Debug log**
Build version: 2.4.3-dev_49ae42b5_2021-11-12 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-14 11:39:39 
Device: Google Pixel 6 Pro 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.IllegalArgumentException: com.health.openscale.dev: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
	at android.app.PendingIntent.checkFlags(PendingIntent.java:375)
	at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:645)
	at android.app.PendingIntent.getBroadcast(PendingIntent.java:632)
	at com.health.openscale.core.alarm.AlarmBackupHandler.getPendingAlarmIntent(AlarmBackupHandler.java:78)
	at com.health.openscale.core.alarm.AlarmBackupHandler.disableAlarm(AlarmBackupHandler.java:84)
	at com.health.openscale.gui.preferences.BackupPreferences.updateBackupPreferences(BackupPreferences.java:94)
	at com.health.openscale.gui.preferences.BackupPreferences.onCreatePreferences(BackupPreferences.java:77)
	at androidx.preference.PreferenceFragmentCompat.onCreate(PreferenceFragmentCompat.java:160)
	at androidx.fragment.app.Fragment.performCreate(Fragment.java:2949)
	at androidx.fragment.app.FragmentStateManager.create(FragmentStateManager.java:475)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:278)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2189)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2106)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:2002)
	at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7839)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)

oliexdev, 2021-11-15:
thanks for reporting should be fixed now in the dev build.

oliexdev, 2021-11-15:
should also implemented in the release version 2.4.4.

closed 2021-11-15 by oliexdev

rodakswe, 2021-11-15:
Can confirm it works in both dev and 2.4.4.
Thanks for fixing :)