oliexdev/openScale #792: Crash when trying to connect to SBF75

Egareg1, 2021-11-15:
**Describe the bug**
After having bought a SBF75 scale, I installed the Open Scale app and try to connect the scale. The scale seems to be recognised as it is listed into Bluetooth device, but the app crash when trying to connect to SBF75. 

**Debug log**
Build version: 2.4.1 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-15 22:12:53 
Device: Wheatek MIX 
OS version: Android 7.0 (SDK 24) 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:228)
		at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1182)
			at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
				at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
					at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.writeBytes(BluetoothBeurerSanitas.java:947)
						at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.sendAlternativeStartCode(BluetoothBeurerSanitas.java:976)
							at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onNextStep(BluetoothBeurerSanitas.java:200)
								at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:598)
									at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:600)
										at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
											at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
												at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
													at android.os.Handler.handleCallback(Handler.java:836)
														at android.os.Handler.dispatchMessage(Handler.java:103)
															at android.os.Looper.loop(Looper.java:203)
																at android.app.ActivityThread.main(ActivityThread.java:6247)
																	at java.lang.reflect.Method.invoke(Native Method)
																		at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
																			at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)

oliexdev, 2021-11-18:
please try out the latest version

Egareg1, 2021-11-19:
Hi, 

Thanks for your reply. 
The new version has just come on fdroid after I wrote my issue. 

I have updated the app on my phone and still have the issue : 
Build version: 2.4.3 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-19 13:52:21 
Device: Wheatek MIX 
OS version: Android 7.0 (SDK 24) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1206)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.writeBytes(BluetoothBeurerSanitas.java:947)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.sendAlternativeStartCode(BluetoothBeurerSanitas.java:976)
	at com.health.openscale.core.bluetooth.BluetoothBeurerSanitas.onNextStep(BluetoothBeurerSanitas.java:200)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
	at android.os.Handler.handleCallback(Handler.java:836)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:203)
	at android.app.ActivityThread.main(ActivityThread.java:6247)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)

How can I proceed to connect the SBF75 scale to the app and download the data ? 

Regards, 
Egareg JAOUEN

oliexdev, 2021-11-21:
Could you post the openScale log file please

deweyjd4, 2021-11-29:
My scale shows as connected in the Bluetooth settings (QN-Scale). When I weigh myself, the app will crash, returning this error. 

Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-28 20:41:06 
Device: Google Pixel 3 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1206)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:209)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7842)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)