oliexdev/openScale #793: Support for Tanita BC-401

MaxParadiz, 2021-11-20:
**Tanita BC-401**
Product page: https://tanita.eu/bc-401/
My Tanita App page: https://play.google.com/store/apps/details?id=com.tanita.mytanita&hl=en&gl=US

**Step 1: Read the general reverse engineer process**

- [ ✓] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**

1. Bluetooth HCI Snoop log file

This is a text file with the summarized data and the assignments:
[SummarizedData.txt](https://github.com/oliexdev/openScale/files/7575478/SummarizedData.txt)

user settings in the vendors app:
```
Male, 180 cm height , age 29, moderate activity level
```
measured true values in the vendors app for the 1. HCI Snoop log file:
```
Date: 20.11.2021
Time: 19:02
Weight: 80.6 kg
BMI: 24.9
Body fat %: 22.2%
Muscle mass: 59.6 kg
Body type: 2
Bone mass: 3.1 kg
Visceral fat: 6
BMR: 1852 kcal
Metabolic age: 39
Body water: 52.9%

```
File:
[btsnoop-1-20211120190207.log](https://github.com/oliexdev/openScale/files/7575474/btsnoop-1-20211120190207.log)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
Male, 180 cm height , age 29, moderate activity level
```
measured true values in the vendors app for the 2. HCI Snoop log file:
```
Date: 20.11.2021
Time: 20:09
Weight:  22.4 kg
BMI: 6.9
Body fat %:  5%
Muscle mass:  20.1 kg
Body type: 7
Bone mass: 1.2 kg
Visceral fat: 1
BMR: 739 kcal
Metabolic age: 14
Body water: 85.0%

```

[btsnoop-2-20211120200923.log](https://github.com/oliexdev/openScale/files/7575477/btsnoop-2-20211120200923.log)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
Male, 180 cm height , age 29, moderate activity level
```
measured true values in the vendors app for the 3. HCI Snoop log file:
```
Date: 20.11.2021
Time: 20:27
Weight:   79.8 kg
BMI:  24.6
Body fat %:  19.7%
Muscle mass:  60.9 kg
Body type: 5
Bone mass: 3.2 kg
Visceral fat: 5
BMR: 1885 kcal
Metabolic age:  31
Body water: 53.1%

```
[btsnoop-3-20211120202718.log](https://github.com/oliexdev/openScale/files/7575475/btsnoop-3-20211120202718.log)

**Step 3: Discover Bluetooth services and characteristic**

OpenScale debug file: 
[openScale_2021-11-20_21-25.txt](https://github.com/oliexdev/openScale/files/7575480/openScale_2021-11-20_21-25.txt)

Bluetoothctl characteristics:
[bluetoothctl.txt](https://github.com/oliexdev/openScale/files/7575482/bluetoothctl.txt)

------
Thank you very much for putting this project together! Looking at how bluetooth works has been an interesting learning experience. I tried to figure this out myself, but I got stuck trying to get the scale to respond using btgatt-client. I can connect the scale to the PC via bluetooth, but the scale will still show the pairing symbol on the screen.  With btgatt-client I connect and use 'write-value 0x0011 1 0', and this does write the value but nothing else happens. I will continue reading and see if I can advance a bit further. Please let me know if there is some other information I can get for you.

Max

oliexdev, 2021-11-21:
thanks for the input. I created the basic structure in openScale, so please try out the [Tanitas branch](https://github.com/oliexdev/openScale/releases/tag/travis-dev_Tanita_BC-401-build) and post the openScale log file.

At first we have to check if we get some data out of the scale. Do you know on which Bluetooth characteristic you send the command? Did you send 0x01?

MaxParadiz, 2021-11-21:
This is OpenScale log file:

[openScale_2021-11-21_13-40.txt](https://github.com/oliexdev/openScale/files/7576368/openScale_2021-11-21_13-40.txt)

The behaviour is the following:

- I hold the button on the scale and the 'pairing symbol' appears. If I do pair the new OpenScale or my computer with the scale, then the pairing symbol will remain for a few minutes, but it won't continue. Eventually it shuts off.

[image]

- I have logged the pairing of the My Tanita app to the scale, and this is the output (I paired, then shut off bluetooth):
[pairing.log](https://github.com/oliexdev/openScale/files/7576381/pairing.log)
 
From Wireshark, this is what I see happens:

(1) Phone sends "Write Request" to 0x0011 of value 01 00
(2) Scale sends a Write Response (0x13)
(3) Phone sends a read request of 0x11
(4) Scale sends a read response (0x0b)
(5) Phone sends a "Write command" to Handle 0x0013 with the value "000002100027000331613032646337352d356162"

I am trying to replicate this using btgatt-client (I have attached the [log here](https://github.com/oliexdev/openScale/files/7576395/btgatt-client.log)), but I am not sure about how to send the write command to handler 0x0013. This is not even listed as a handler, and I have no permission to write to it. I can only write to 0x0c, 0x11,  and 0x16. I can register notifications for 0x10 and 0x13. I do not yet understand how this works... I also tried running 'write-value 0x11 0x01', 'write-value 0x0c 0x01', and 'write-value 0x16 0x01'. It seems that writing any value to any of these three writes the value to all of them.

- Every second time I try to pair to to my computer, an error occurs. In the computer the error says: Connecting to device... Failed to connect: Software caused connection abort. Maybe this is relevant

oliexdev, 2021-11-21:
Now you have a device which call TNT_BW?!? In your first log file it called TNT_PAIR. Normally, you don't need to pair to a scale but here it could be an exception.

Additionally, it seems you logged in openScale the device discovery under Bluetooth->settings. Please enable the openScale log file while you try to connect to the scale (top right Bluetooth symbol) and post the result here.

Thanks in advance.

MaxParadiz, 2021-11-21:
Ah, sorry, here is the log using the top right Bluetooth symbol:

[openScale_2021-11-21_16-55.txt](https://github.com/oliexdev/openScale/files/7576573/openScale_2021-11-21_16-55.txt)

I also get the upside-down ladybug with the following error:

[bug_report.log](https://github.com/oliexdev/openScale/files/7576589/bug_report.log)

The name changes between three different names: TNT_PAIR, TNT_BW, and BC-401EU. 

If I press the pair pair the devices shows up as TNT_Pair, but if the scale is off and I try to connect to it it will connect and alternate between TNT_BW and BC-401EU.  If I pair to it through the app while the scale is off, it will turn on. 

Here is the bluetoothctl log example, I connect to the same MAC address three times in a row (first two the scale is off, third time the scale is in pairing mode) and different names show up:
[names.log](https://github.com/oliexdev/openScale/files/7576581/names.log)

MaxParadiz, 2021-11-21:
Here is the openScale log of what happens if I connect when the scale is off. The connection will be established, and the name changes to TNT_BW, but the scale will not turn on and the connection drops. I have tried a few things, like connecting and then turning the scale on, but it won't work.

[openScale_2021-11-21_17-11.txt](https://github.com/oliexdev/openScale/files/7576599/openScale_2021-11-21_17-11.txt)

oliexdev, 2021-11-22:
it seems it's more complex, as I don't own this scale it's hard to analyze it. Could you get the Bluetooth characteristic from TNT_BW? Do you have to pair the device with the original app?

MaxParadiz, 2021-11-22:
One does need to go through a short pairing procedure the first time that they want to connect a device to the app. It is also obligatory to create an account before this option is even available. This is what the pairing looks like in the app:

[image]

After some testing, this is what I think:

TNT_Pair: Name while handling the pairing
TNT_BW: The name when you connect to the scale when it is turned off
BC-401EU: This is the name of the connected scale

TNT_PAIR, TNT_BW, and BC-401EU have the same services and characteristics. Only the name and alias appears to change.

[Bluetoothctl characteristics](https://github.com/oliexdev/openScale/files/7580530/characteristics.log)

This is the behavior that I observe when connecting via the terminal using Bluetoothctl:

- If the scale is off, and I connect to it, I will alternately pair to TNT_BW or BC-401EU, and the name will change as soon as I connect. The connection times out almost immediately.

Example of connecting several times in a row while the scale is off:

`[bluetooth]# connect BC:82:5D:0D:7F:E0
Attempting to connect to BC:82:5D:0D:7F:E0
[CHG] Device BC:82:5D:0D:7F:E0 Connected: yes
Connection successful
[CHG] Device BC:82:5D:0D:7F:E0 Name: BC-401EU
[CHG] Device BC:82:5D:0D:7F:E0 Alias: BC-401EU
[CHG] Device BC:82:5D:0D:7F:E0 Connected: no
[bluetooth]# connect BC:82:5D:0D:7F:E0
Attempting to connect to BC:82:5D:0D:7F:E0
[CHG] Device BC:82:5D:0D:7F:E0 Connected: yes
[CHG] Device BC:82:5D:0D:7F:E0 Name: TNT_BW
[CHG] Device BC:82:5D:0D:7F:E0 Alias: TNT_BW
Connection successful
[CHG] Device BC:82:5D:0D:7F:E0 Name: BC-401EU
[CHG] Device BC:82:5D:0D:7F:E0 Alias: BC-401EU
[CHG] Device BC:82:5D:0D:7F:E0 Connected: no
[bluetooth]# connect BC:82:5D:0D:7F:E0
Attempting to connect to BC:82:5D:0D:7F:E0
[CHG] Device BC:82:5D:0D:7F:E0 Connected: yes
[CHG] Device BC:82:5D:0D:7F:E0 Name: TNT_BW
[CHG] Device BC:82:5D:0D:7F:E0 Alias: TNT_BW
Connection successful
[CHG] Device BC:82:5D:0D:7F:E0 Connected: no
[CHG] Controller 00:1A:7D:DA:71:13 Class: 0x00000000
[CHG] Controller 00:1A:7D:DA:71:13 Powered: no
[CHG] Controller 00:1A:7D:DA:71:13 Discovering: no
[CHG] Controller 00:1A:7D:DA:71:13 Class: 0x006c0104
[CHG] Controller 00:1A:7D:DA:71:13 Powered: yes
`

-The only way that I have managed to establish a stable connection with BC-401EU is to forget the device, pair to TNT_Pair (which appears if I press on "Set Pairing"), and then connecting to the device, like this:

[Pair-BC401EU.log](https://github.com/oliexdev/openScale/files/7581149/Pair-BC401EU.log)

I have tried writing to each of the characteristics when the name is BC-401EU, but I have not managed to figure out what command to write to make the scale react. I have just begun learning about characteristics and how to write to them, so there can be a very 'obvious' step that I am missing. 

Thank you very much for your assistance with this!

pepijndevos, 2022-02-25:
We recently got this scale and troubles with the official app lead me to this issue. If my girlfriend can't get the app to work maybe I'll try to play around with the info here. (basically she set it up, but the second time she used it it would not connect and after removing the scale from the app there seems no way to add it again wtf)