oliexdev/openScale #794: Can not connect to scale Beurer BF710 with new phone samsung SM-G781B Android 11 (works on old phone)

CRISPRCUT, 2021-11-21:
**Describe the bug**
Can not connect to scale Beurer BF710 with new phone samsung SM-G781B Android 11
BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>

No problem with old phone samsung SM-G950F Android 9
 BluetoothCommunication: SUCCESS: Writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>

**To Reproduce**
Steps to reproduce the behavior:
1. start openScale
2. click on Bluetooth
3. Step on scale
no data transphered with SM-G781B (works with SM-G950F)

Reproduced with openScale v2.4.5 (60)

**Expected behavior**
connect to scale with SM-G781B and request saved measurements

**Additional context**
i also tryed different versions on both phones:
same problem with v2.4.5-pro (60) on  SM-G781B does not work
same problem with v2.3.2-dev (60) on  SM-G781B does not work
v2.4.1-pro (56) on SM-G950F works

**Debug log**
log from new phone that does not work:

2021-11-21 11:02:16.680 Debug [2] AboutPreferences: Debug log enabled, openScale v2.4.5 (60), SDK 30, samsung SM-G781B
2021-11-21 11:02:16.681 Debug [2] AboutPreferences: Selected user id(1) name(Ralf) birthday(Thu Sep 01 00:00:00 GMT+01:00 1976) age(42) body height(180,00) scale unit(kg) gender(male) initial weight(78,00) goal weight(75,00) goal date(Sat May 21 10:49:33 GMT+02:00 2022) measure unt(cm) activity level(0) assisted weighing(false)
2021-11-21 11:03:29.609 Debug [2] OpenScale: Trying to connect to bluetooth device [D4:F5:13:5B:F1:47] (Beurer BF710)
2021-11-21 11:03:29.611 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2021-11-21 11:03:29.621 Info [2] BluetoothCentralManager: scan started
2021-11-21 11:03:29.622 Debug [2] BluetoothCommunication: Stop machine state
2021-11-21 11:03:33.212 Debug [2] BluetoothCommunication: Found peripheral 'Beurer BF710'
2021-11-21 11:03:33.214 Info [2] BluetoothCentralManager: scan stopped
2021-11-21 11:03:34.217 Debug [2] BluetoothCommunication: Try to connect to BLE device D4:F5:13:5B:F1:47
2021-11-21 11:03:34.322 Info [2] BluetoothPeripheral: connect to 'Beurer BF710' (D4:F5:13:5B:F1:47) using transport LE
2021-11-21 11:03:34.326 Info [2] BluetoothPeripheral: peripheral 'D4:F5:13:5B:F1:47' is connecting
2021-11-21 11:03:34.654 Info [2274] BluetoothPeripheral: connected to 'Beurer BF710' (NONE) in 0,3s
2021-11-21 11:03:34.656 Debug [2] BluetoothPeripheral: discovering services of 'Beurer BF710' with delay of 0 ms
2021-11-21 11:03:35.092 Debug [2274] BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2021-11-21 11:03:35.744 Info [2274] BluetoothPeripheral: discovered 5 services for 'Beurer BF710'
2021-11-21 11:03:35.748 Debug [2] BluetoothCommunication: connected to 'Beurer BF710'
2021-11-21 11:03:35.748 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2021-11-21 11:03:35.749 Debug [2] BluetoothCommunication: Resume machine state
2021-11-21 11:03:35.749 Debug [2] BluetoothCommunication: Step Nr 0
2021-11-21 11:03:35.750 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2021-11-21 11:03:35.751 Debug [2] BluetoothCommunication: Stop machine state
2021-11-21 11:03:35.766 Debug [2] MainActivity: Bluetooth connection successful established
2021-11-21 11:03:35.792 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2021-11-21 11:03:35.792 Debug [2] BluetoothCommunication: Resume machine state
2021-11-21 11:03:35.793 Debug [2] BluetoothCommunication: Step Nr 1
2021-11-21 11:03:35.793 Debug [2] BluetoothBeurerSanitas: Sending command: ID_START_NIBBLE_INIT
2021-11-21 11:03:35.795 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2021-11-21 11:03:35.796 Debug [2] BluetoothCommunication: Stop machine state
2021-11-21 11:03:35.797 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:03:35.818 Debug [2274] BluetoothPeripheral: connection parameters: interval=45.0ms latency=0 timeout=5s
2021-11-21 11:03:40.872 Error [2274] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'ERROR'
2021-11-21 11:03:40.874 Error [2] BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:03:40.878 Info [2274] BluetoothPeripheral: peripheral 'Beurer BF710' disconnected with status 'CONNECTION_TIMEOUT' (8)
2021-11-21 11:03:40.892 Debug [2] BluetoothCommunication: disconnected 'Beurer BF710' with status 8
2021-11-21 11:04:35.773 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-11-21 11:04:35.776 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-11-21 11:04:35.777 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2021-11-21 11:04:35.778 Error [2] BluetoothCentralManager: cannot cancel connection to unknown peripheral D4:F5:13:5B:F1:47
2021-11-21 11:04:35.803 Debug [2] MainActivity: Bluetooth connection successful disconnected

log from old phone that works:

[... 91 lines cut ...]
2021-11-21 11:07:08.364 Debug [2] BluetoothBeurerSanitas: Measure command successfully received
2021-11-21 11:07:09.390 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 01] on 0xffe1
2021-11-21 11:07:09.427 Debug [2] BluetoothPeripheral: writing <e7f1590301> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:09.531 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:09.535 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 02] on 0xffe1
2021-11-21 11:07:09.542 Debug [2] BluetoothPeripheral: writing <e7f1590302> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:09.629 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 02> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:09.634 Debug [2] BluetoothCommunication: Invoke write bytes [E7 F1 59 03 03] on 0xffe1
2021-11-21 11:07:09.640 Debug [2] BluetoothCommunication: Invoke write bytes [E7 43 00 00 00 00 00 00 00 66] on 0xffe1
2021-11-21 11:07:09.671 Debug [2] OpenScale: Added measurement: ID: 0, USER_ID: 1, DATE_TIME: Sun Nov 21 11:07:09 GMT+01:00 2021, WEIGHT: 79,35, FAT: 18,00, WATER: 57,50, MUSCLE: 42,30, LBM: 0,00, WAIST: 0,00, HIP: 0,00, BONE: 11,55, CHEST: 0,00, THIGH: 0,00, ARM: 0,00, NECK: 0,00, CALIPER1: 0,00, CALIPER2: 0,00, CALIPER3: 0,00, COMMENT: 
2021-11-21 11:07:09.686 Debug [2] BluetoothPeripheral: writing <e7f1590303> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:10.047 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 F1 59 03 03> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:10.051 Debug [2] BluetoothPeripheral: writing <e7430000000000000066> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:10.116 Debug [2] BluetoothCommunication: SUCCESS: Writing <E7 43 00 00 00 00 00 00 00 66> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-21 11:07:10.187 Debug [2] BluetoothBeurerSanitas: Saved measurements successfully deleted for user RAL
2021-11-21 11:07:10.189 Debug [2] BluetoothCommunication: Resume machine state
2021-11-21 11:07:10.192 Debug [2] BluetoothCommunication: Step Nr 9
2021-11-21 11:07:10.194 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2021-11-21 11:07:57.742 Info [9359] BluetoothPeripheral: peripheral 'Beurer BF710' disconnected with status 'GATT CONN TIMEOUT'
2021-11-21 11:07:57.827 Debug [2] BluetoothCommunication: disconnected 'Beurer BF710' with status 8
2021-11-21 11:08:10.227 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect
2021-11-21 11:08:10.240 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-11-21 11:08:10.242 Info [2] BluetoothCentral: no scan to stop because no scan is running
2021-11-21 11:08:10.246 Error [2] BluetoothCentral: cannot cancel connection to unknown peripheral D4:F5:13:5B:F1:47
2021-11-21 11:08:10.406 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2021-11-21:
Does it work if you update the openScale version to 2.4.5 on your SM-G950F?

CRISPRCUT, 2021-11-21:
checked now with 2.4.5-pro on SM-G950F and it works

oliexdev, 2021-11-22:
and on the SM-G781B device the 2.4.5 still doesn't work? Did you check the permissions if they were all granted?

CRISPRCUT, 2021-11-23:
Still does not work on  SM-G781B and 2.4.5-pro. I checked the permissions - all set.

log is almost the same
2021-11-23 15:16:41.393 Debug [2] OpenScale: Disconnecting from bluetooth device
replaces 
2021-11-21 11:04:35.773 Debug [2] BluetoothCommunication: Timeout Bluetooth disconnect

full log:

2021-11-23 15:16:05.014 Debug [2] AboutPreferences: Debug log enabled, openScale v2.4.5-pro (60), SDK 30, samsung SM-G781B
2021-11-23 15:16:05.016 Debug [2] AboutPreferences: Selected user id(1) name(Ralf) birthday(Thu Sep 01 00:00:00 GMT+01:00 1976) age(42) body height(180,00) scale unit(kg) gender(male) initial weight(76,00) goal weight(74,00) goal date(Wed Oct 02 10:18:41 GMT+02:00 2019) measure unt(cm) activity level(0) assisted weighing(false)
2021-11-23 15:16:13.126 Debug [2] OpenScale: Disconnecting from bluetooth device
2021-11-23 15:16:13.127 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-11-23 15:16:13.128 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2021-11-23 15:16:13.129 Error [2] BluetoothCentralManager: cannot cancel connection to unknown peripheral D4:F5:13:5B:F1:47
2021-11-23 15:16:15.883 Debug [2] OpenScale: Trying to connect to bluetooth device [D4:F5:13:5B:F1:47] (Beurer BF710)
2021-11-23 15:16:15.885 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2021-11-23 15:16:15.897 Info [2] BluetoothCentralManager: scan started
2021-11-23 15:16:15.897 Debug [2] BluetoothCommunication: Stop machine state
2021-11-23 15:16:18.397 Debug [2] BluetoothCommunication: Found peripheral 'Beurer BF710'
2021-11-23 15:16:18.401 Info [2] BluetoothCentralManager: scan stopped
2021-11-23 15:16:19.403 Debug [2] BluetoothCommunication: Try to connect to BLE device D4:F5:13:5B:F1:47
2021-11-23 15:16:19.510 Info [2] BluetoothPeripheral: connect to 'Beurer BF710' (D4:F5:13:5B:F1:47) using transport LE
2021-11-23 15:16:19.514 Info [2] BluetoothPeripheral: peripheral 'D4:F5:13:5B:F1:47' is connecting
2021-11-23 15:16:22.132 Info [13095] BluetoothPeripheral: connected to 'Beurer BF710' (NONE) in 2,6s
2021-11-23 15:16:22.136 Debug [2] BluetoothPeripheral: discovering services of 'Beurer BF710' with delay of 0 ms
2021-11-23 15:16:22.563 Debug [13095] BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2021-11-23 15:16:23.167 Info [13095] BluetoothPeripheral: discovered 5 services for 'Beurer BF710'
2021-11-23 15:16:23.168 Debug [2] BluetoothCommunication: connected to 'Beurer BF710'
2021-11-23 15:16:23.169 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2021-11-23 15:16:23.169 Debug [2] BluetoothCommunication: Resume machine state
2021-11-23 15:16:23.169 Debug [2] BluetoothCommunication: Step Nr 0
2021-11-23 15:16:23.169 Debug [2] BluetoothCommunication: Invoke set notification on 0xffe1
2021-11-23 15:16:23.169 Debug [2] BluetoothCommunication: Stop machine state
2021-11-23 15:16:23.177 Debug [2] MainActivity: Bluetooth connection successful established
2021-11-23 15:16:23.200 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000ffe1-0000-1000-8000-00805f9b34fb
2021-11-23 15:16:23.200 Debug [2] BluetoothCommunication: Resume machine state
2021-11-23 15:16:23.200 Debug [2] BluetoothCommunication: Step Nr 1
2021-11-23 15:16:23.201 Debug [2] BluetoothBeurerSanitas: Sending command: ID_START_NIBBLE_INIT
2021-11-23 15:16:23.201 Debug [2] BluetoothCommunication: Invoke write bytes [E6 01] on 0xffe1
2021-11-23 15:16:23.202 Debug [2] BluetoothCommunication: Stop machine state
2021-11-23 15:16:23.203 Debug [2] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-23 15:16:23.250 Debug [13095] BluetoothPeripheral: connection parameters: interval=45.0ms latency=0 timeout=5s
2021-11-23 15:16:28.291 Error [13095] BluetoothPeripheral: writing <e601> to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'ERROR'
2021-11-23 15:16:28.292 Error [2] BluetoothCommunication: ERROR: Failed writing <E6 01> to <0000ffe1-0000-1000-8000-00805f9b34fb>
2021-11-23 15:16:28.294 Info [13095] BluetoothPeripheral: peripheral 'Beurer BF710' disconnected with status 'CONNECTION_TIMEOUT' (8)
2021-11-23 15:16:28.308 Debug [2] BluetoothCommunication: disconnected 'Beurer BF710' with status 8
2021-11-23 15:16:41.393 Debug [2] OpenScale: Disconnecting from bluetooth device
2021-11-23 15:16:41.394 Debug [2] BluetoothCommunication: Bluetooth disconnect
2021-11-23 15:16:41.394 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2021-11-23 15:16:41.395 Error [2] BluetoothCentralManager: cannot cancel connection to unknown peripheral D4:F5:13:5B:F1:47
2021-11-23 15:16:41.406 Debug [2] MainActivity: Bluetooth connection successful disconnected

oliexdev, 2021-11-24:
sorry I don't really know why its works on your SM-G950F and not on your  SM-G781B  with the same openScale version.
Somehow the Bluetooth write command failed on your SM-G781B device failed.

```2021-11-23 15:16:28.291 Error [13095] BluetoothPeripheral: writing to characteristic <0000ffe1-0000-1000-8000-00805f9b34fb> failed, status 'ERROR'```

Maybe @weliem from blessed-android have some hints!?

weliem, 2021-11-24:
Add a delay of 300ms before and after this write. This scale needs some time to 'wake up'...

oliexdev, 2021-11-28:
I was wondering why is it working on the other device?

closed 2022-04-15 by oliexdev