oliexdev/openScale #795: Feature request - Support Scale connection through MQTT

acabrol, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
BF850 has hardware limitation about BLE hosts generation. 
Connection with new android hardware phone raise error during weighing.
https://www.beurer.com/web/gb/in-focus/connect/system-requirements.php
Too few mobile supported.
`    Redmi Note 7
    Huawei P9
    Huawei P20 Pro
    LG Google Nexus 5
    Samsung Galaxy Note 8
    Samsung Galaxy J5
    Samsung Galaxy J7
    Samsung Galaxy A3
    Samsung Galaxy A5
    Samsung Galaxy S4
    Samsung Galaxy S5
    Samsung Galaxy S6
    Samsung Galaxy S7
    Samsung Galaxy S8
    Samsung Galaxy S9
    Samsung Galaxy S10`

**Describe the solution you'd like**
Use [ESP32-BLE2MQTT ](https://github.com/shmuelzon/esp32-ble2mqtt) Project to connect OpenScale to the BF850 through an MQTT server.

**Describe alternatives you've considered**
- Install Official App Health manager => same error during weighing
- Try other proprietary or open source apps => same error during weighing
- Try to find a way to update the BF850 firmware => no solution

**Additional context**
This could be an opportunity to automatically synchronise weigh data without having to connect the scale to the phone (through lan or internet).
Increase integration with MQTT or domotic servers

acabrol, 2021-11-23:
I posted a crowd funding call for this feature here :
https://www.bountysource.com/issues/103484231-feature-request-support-scale-connection-through-mqtt

oliexdev, 2021-11-24:
use [openScale sync](https://github.com/oliexdev/openScale/wiki/MQTT-support).

closed 2021-12-12 by oliexdev