oliexdev/openScale #796: Error message in OpenScale on Samsung Galaxy Grand Prime and BF70

Consuelo30, 2021-11-24:
Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-23 22:18:56 
Device: Samsung SM-G530FZ 
OS version: Android 7.1.2 (SDK 25) 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'int android.widget.AbsListView.getCheckedItemPosition()' on a null object reference
at com.health.openscale.gui.MainActivity$10.onClick(MainActivity.java:487)
at com.android.internal.app.AlertController$ButtonHandler.handleMessage(AlertController.java:166)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loop(Looper.java:154)
at android.app.ActivityThread.main(ActivityThread.java:6236)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:891)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:781)

oliexdev, 2021-11-28:
What did you do to get this error message?

Consuelo30, 2021-11-29:
This message appears when trying to add a new user, by tapping the plus
sign and then "OK" in the dialog. The smartphone doesn't show any data
about weight or other parameters, but asks me for permission to activate
Bluetooth.

Thanks for your answer

closed 2022-04-15 by oliexdev