oliexdev/openScale #801: Crashing when trying to add a Bluetooth scale on Samsung Galaxy A22 5G

Barthelemy-Drabczuk, 2021-11-27:
**Describe the bug**
When trying to add a scale in the menu, the app shows up a crash screen, telling to report it here. My device is a Samsung Galaxy A22 5G running Android 11.

**To Reproduce**
Steps to reproduce the behavior:
1. Go in the sidebar menu
2. Select "Parameters"
3. Select "Bluetooth"
4. Finally, when clicking on "Your Bluetooth scale" the app crashes

Reproduced with version v2.4.5 (60)

**Expected behavior**
I was expecting to be able to select the right Bluetooth driver for my scale.

**Additional context**
My phone default language is in French (I have no idea if it's relevant).

**Debug log**
Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-11-27 20:37:44 
Device: Samsung SM-A226B 
OS version: Android 11 (SDK 30) 
 
Stack trace:
java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
    at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:875)
    at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:835)
    at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:401)
    at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
    at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
    at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
    at androidx.fragment.app.Fragment.performResume(Fragment.java:3039)
    at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:607)
    at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:306)
    at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:745)
    at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:672)
    at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:776)
    at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:147)
    at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:120)
    at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:297)
    at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2202)
    at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2106)
    at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:2002)
    at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
    at android.os.Handler.handleCallback(Handler.java:938)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loop(Looper.java:246)
    at android.app.ActivityThread.main(ActivityThread.java:8633)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)

oliexdev, 2021-11-28:
please grant the location permission.

oliexdev, 2021-11-28:
and enable your GPS on your device

closed 2021-11-28 by oliexdev

Barthelemy-Drabczuk, 2021-12-01:
It worked, thanks !