oliexdev/openScale #810: Connecting Bluetooth crashes the app

pkscwc, 2021-12-19:
Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-12-19 11:44:42 
Device: Xiaomi M2004J19PI 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.IllegalArgumentException: characteristic <0000fff1-0000-1000-8000-00805f9b34fb> does not support writeType 'WITH_RESPONSE'
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1225)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothHesley.onNextStep(BluetoothHesley.java:49)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onNotificationStateUpdate(BluetoothCommunication.java:461)
	at com.welie.blessed.BluetoothPeripheral$1$2.run(BluetoothPeripheral.java:238)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:236)
	at android.app.ActivityThread.main(ActivityThread.java:7864)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:620)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1011)

mementomori74, 2021-12-21:
i had the same issue. one plus nord

Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-12-20 21:59:02 
Device: OnePlus EB2103 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:875)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:835)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:401)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3039)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:607)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:306)
	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:745)
	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:672)
	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:776)
	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:147)
	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:120)
	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:297)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2202)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2106)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:2002)
	at androidx.fragment.app.FragmentManager$5.run(FragmentManager.java:524)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:233)
	at android.app.ActivityThread.main(ActivityThread.java:8063)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:631)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:978)

Stonerocker, 2021-12-29:
looks like I have the same issue.

to reproduce:
1. add scale in Bluetooth settings
2. create a new user
3. use scale and connect Bluetooth
after a few seconds a list of radio buttons appears and instantly the app crashes.

Build version: 2.4.5 
Build date: 1981-01-01 01:01:02 
Current date: 2021-12-29 21:00:46 
Device: Samsung SM-J530F 
OS version: Android 9 (SDK 28) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1206)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeTargetWeight(BluetoothBeurerBF105.java:132)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeUserDataToScale(BluetoothBeurerBF105.java:72)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep(BluetoothStandardWeightProfile.java:195)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.handleUserControlPointNotify(BluetoothStandardWeightProfile.java:279)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onBluetoothNotify(BluetoothStandardWeightProfile.java:250)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.onBluetoothNotify(BluetoothBeurerBF105.java:82)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:482)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:276)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:216)
	at android.app.ActivityThread.main(ActivityThread.java:7266)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975

Stonerocker, 2021-12-30:
I switched to v2.4.3. Now Bluetooth data transfer works again.

Stonerocker, 2022-01-03:
dev version works similar to v2.4.3. However, it crashes on first attempt to connect to Bluetooth device.

ivasemma-6872, 2022-05-29:
Dear oliexdev,

First of all, thank you for your app ! Very good !

But, I am also facing a bug with openScale _(downloaded from F-Droid)_ when triying to add a Bluetooth device: it crashes.

Thank you very much.

Best regards,

`Build version: 2.4.6`
`Build date: 1981-01-01 01:01:02 `
`Current date: 2022-05-29 08:59:02 `
`Device: Blackview A7 `
`OS version: Android 7.0 (SDK 24) `
` `
`Stack trace:  `
`java.lang.SecurityException: app does not have ACCESS_COARSE_LOCATION permission, cannot start scan`
`	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:882)`
`	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)`
`	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)`
`	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)`
`	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)`
`	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)`
`	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)`
`	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)`
`	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)`
`	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:771)`
`	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:690)`
`	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:855)`
`	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:154)`
`	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:123)`
`	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:306)`
`	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1853)`
`	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1764)`
`	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1701)`
`	at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:488)`
`	at android.os.Handler.handleCallback(Handler.java:836)`
`	at android.os.Handler.dispatchMessage(Handler.java:103)`
`	at android.os.Looper.loop(Looper.java:203)`
`	at android.app.ActivityThread.main(ActivityThread.java:6251)`
`	at java.lang.reflect.Method.invoke(Native Method)`
`	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)`
`	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)`

oliexdev, 2022-05-29:
@ivasemma-6872 you need to grant the FINE LOCATION permission to search for Bluetooth devices.

closed 2022-12-08 by oliexdev