oliexdev/openScale #816: Auto backup works only to the Download or Documents folders

marvinwankersteen, 2022-01-09:
**Describe the bug**
Hello,
for me, the autobackup function has not worked properly since mid 2021. I think since I switched to Android 11.
I have specified `Transfer/Backup/openScale` as the backup folder. That had also worked with it. I was able to look into it now and see where the problem is and it seems that openScale is only allowed to write to the `Download` and `Documents`.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings' --> 'Backup'
2. Click on 'Export folder' and set `Transfer/Backup/openScale` (`Transfer` will be synced with syncthing)
3. See error in logfile

Reproduced with: v2.4.5 (60) from F-Droid

**Expected behavior**
Database should be saved.

**Additional context**
Is this the normal behavior, that only the backup may be written to `Download` or `Documents`?
Because if I manually export the backup to `Transfer/Backup/openScale` it works. Only the automatic backup does not work.
If I specify as backup folder for example `Download`, the backup is written successfully:
```
2022-01-09 12:50:10.075 Debug [2] AlarmBackupHandler: openScale Auto Backup to /storage/emulated/0/Download/2022-01-09_openScale.db
```

Logcat:
```
01-09 12:42:07.188  6494 24285 W MediaProvider: Forgot to handle a top level directory in getContentUriForFile?
01-09 12:42:07.191  6494 24285 E MediaProvider: insertFileIfNecessary failed
01-09 12:42:07.191  6494 24285 E MediaProvider: java.lang.IllegalArgumentException: Primary directory Transfer not allowed for content://media/external_primary/file; allowed directories are [Download, Documents]
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.ensureFileColumns(MediaProvider.java:2707)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.ensureUniqueFileColumns(MediaProvider.java:2372)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.insertFile(MediaProvider.java:2981)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.insertInternal(MediaProvider.java:3533)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.insert(MediaProvider.java:3249)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.insertFileForFuse(MediaProvider.java:6717)
01-09 12:42:07.191  6494 24285 E MediaProvider: 	at com.android.providers.media.MediaProvider.insertFileIfNecessaryForFuse(MediaProvider.java:6804)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: Error while exporting database
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: java.io.FileNotFoundException: open failed: EPERM (Operation not permitted)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.os.ParcelFileDescriptor.openInternal(ParcelFileDescriptor.java:344)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.os.ParcelFileDescriptor.open(ParcelFileDescriptor.java:231)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1793)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1518)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1494)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.health.openscale.core.OpenScale.copyFile(OpenScale.java:478)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.health.openscale.core.OpenScale.exportDatabase(OpenScale.java:473)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.health.openscale.core.alarm.AlarmBackupHandler.executeBackup(AlarmBackupHandler.java:113)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.health.openscale.core.alarm.ReminderBootReceiver.handleBackupAlarm(ReminderBootReceiver.java:44)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.health.openscale.core.alarm.ReminderBootReceiver.onReceive(ReminderBootReceiver.java:30)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.app.ActivityThread.handleReceiver(ActivityThread.java:4026)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.app.ActivityThread.access$1400(ActivityThread.java:237)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1924)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.os.Handler.dispatchMessage(Handler.java:106)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.os.Looper.loop(Looper.java:223)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at android.app.ActivityThread.main(ActivityThread.java:7664)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at java.lang.reflect.Method.invoke(Native Method)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
01-09 12:42:07.192 25415 25415 E AlarmBackupHandler: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

**Debug log**
```
2022-01-09 12:37:05.595 Error [2] AlarmBackupHandler: Error while exporting database
java.io.FileNotFoundException: open failed: EPERM (Operation not permitted)
	at android.os.ParcelFileDescriptor.openInternal(ParcelFileDescriptor.java:344)
	at android.os.ParcelFileDescriptor.open(ParcelFileDescriptor.java:231)
	at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1793)
	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1518)
	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1494)
	at com.health.openscale.core.OpenScale.copyFile(OpenScale.java:478)
	at com.health.openscale.core.OpenScale.exportDatabase(OpenScale.java:473)
	at com.health.openscale.core.alarm.AlarmBackupHandler.executeBackup(AlarmBackupHandler.java:113)
	at com.health.openscale.core.alarm.ReminderBootReceiver.handleBackupAlarm(ReminderBootReceiver.java:44)
	at com.health.openscale.core.alarm.ReminderBootReceiver.onReceive(ReminderBootReceiver.java:30)
	at android.app.ActivityThread.handleReceiver(ActivityThread.java:4026)
	at android.app.ActivityThread.access$1400(ActivityThread.java:237)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1924)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

Access to files is allowed.
[image]

momo42, 2022-02-04:
I didn't record logs but can confirm marvins observations. Manual backup works, auto backup doesn't.
Also I'd like to point to this old issue https://github.com/oliexdev/openScale/issues/684 .
Having an 'test auto backup' button would indeed have made things a bit easier. ;)

oliexdev, 2022-04-15:
That's true on Android > 11 they change the storage permission, see https://developer.android.com/about/versions/11/privacy/storage

I will have to change it accordingly if I have some time or somebody could send a PR with this fix.

oliexdev, 2022-04-15:
I did a fix with https://github.com/oliexdev/openScale/commit/d022cd321ccd6cdbeaa463591b97c6aadf50bba9 please try out the latest dev build version of openScale

closed 2022-05-08 by oliexdev

pho3nixx, 2022-05-08:
Sorry, the issue still occurs with v2.4.6 (61) on Android 11 (MIUI 12). When trying to tick the "automatic backup" checkbox, the message "access denied" appears and the box stays unchecked. The only message in debug log is
`2022-05-08 23:27:51.609 Debug [2] AlarmBackupHandler: openScale Auto Backup to /storage/emulated/0/Android/data/com.health.openscale/files/Datensicherung/Openscale/openScale.db`

foss-, 2023-01-12:
Issue is persisting with 2.4.6.for me as well. Should this issue here be re-opened?