oliexdev/openScale #819: Crash when updating android new version

Baandiras, 2022-01-16:
Build version: 2.4.5-pro 
Build date: 1981-01-01 01:01:02 
Current date: 2022-01-16 12:02:19 
Device: Samsung SM-G973F 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{c83d52b 16803:com.health.openscale.pro/u0a245} (pid=16803, uid=10245) requires android.permission.BLUETOOTH_CONNECT
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2437)
	at android.os.Parcel.createException(Parcel.java:2421)
	at android.os.Parcel.readException(Parcel.java:2404)
	at android.os.Parcel.readException(Parcel.java:2346)
	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2897)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1743)
	at android.app.Activity.startActivityForResult(Activity.java:5465)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:597)
	at android.app.Activity.startActivityForResult(Activity.java:5423)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:583)
	at com.health.openscale.gui.MainActivity.invokeConnectToBluetoothDevice(MainActivity.java:584)
	at com.health.openscale.gui.MainActivity.onCreateOptionsMenu(MainActivity.java:535)
	at android.app.Activity.onCreatePanelMenu(Activity.java:4338)
	at androidx.fragment.app.FragmentActivity.onCreatePanelMenu(FragmentActivity.java:287)
	at androidx.appcompat.view.WindowCallbackWrapper.onCreatePanelMenu(WindowCallbackWrapper.java:94)
	at androidx.appcompat.app.AppCompatDelegateImpl$AppCompatWindowCallback.onCreatePanelMenu(AppCompatDelegateImpl.java:3084)
	at androidx.appcompat.view.WindowCallbackWrapper.onCreatePanelMenu(WindowCallbackWrapper.java:94)
	at androidx.appcompat.app.ToolbarActionBar.populateOptionsMenu(ToolbarActionBar.java:456)
	at androidx.appcompat.app.ToolbarActionBar$1.run(ToolbarActionBar.java:57)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8633)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1133)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1335)
	at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:1252)
	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:883)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1871)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1742)

oliexdev, 2022-04-15:
Did you disable the Bluetooth on you phone?

Damjan94, 2022-04-28:
I have a similar issue
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-04-28 10:49:00 
Device: Samsung SM-G935F 
OS version: Android 12 (SDK 32) 
 
Stack trace:  
java.lang.RuntimeException: Unable to resume activity {com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{954404 21726:com.health.openscale/u0a142} (pid=21726, uid=10142) requires android.permission.BLUETOOTH_CONNECT
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4824)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4857)
	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:54)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2253)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7870)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)
Caused by: java.lang.SecurityException: Permission Denial: starting Intent { act=android.bluetooth.adapter.action.REQUEST_ENABLE cmp=com.android.settings/.bluetooth.RequestPermissionActivity } from ProcessRecord{954404 21726:com.health.openscale/u0a142} (pid=21726, uid=10142) requires android.permission.BLUETOOTH_CONNECT
	at android.os.Parcel.createExceptionOrNull(Parcel.java:2425)
	at android.os.Parcel.createException(Parcel.java:2409)
	at android.os.Parcel.readException(Parcel.java:2392)
	at android.os.Parcel.readException(Parcel.java:2334)
	at android.app.IActivityTaskManager$Stub$Proxy.startActivity(IActivityTaskManager.java:2326)
	at android.app.Instrumentation.execStartActivity(Instrumentation.java:1758)
	at android.app.Activity.startActivityForResult(Activity.java:5407)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:597)
	at android.app.Activity.startActivityForResult(Activity.java:5365)
	at androidx.activity.ComponentActivity.startActivityForResult(ComponentActivity.java:583)
	at com.health.openscale.gui.utils.PermissionHelper.requestBluetoothPermission(PermissionHelper.java:56)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:102)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3080)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.FragmentController.dispatchResume(FragmentController.java:284)
	at androidx.fragment.app.FragmentActivity.onResumeFragments(FragmentActivity.java:434)
	at androidx.fragment.app.FragmentActivity.onPostResume(FragmentActivity.java:423)
	at androidx.appcompat.app.AppCompatActivity.onPostResume(AppCompatActivity.java:240)
	at android.app.Activity.performResume(Activity.java:8219)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4814)
	... 13 more
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.wm.ActivityTaskSupervisor.checkStartAnyActivityPermission(ActivityTaskSupervisor.java:1059)
	at com.android.server.wm.ActivityStarter.executeRequest(ActivityStarter.java:983)
	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:672)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1252)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1224)

```

And indeed, my bluetooth was off.

oliexdev, 2022-05-08:
you have to turn on the Bluetooth if you want to use it.

closed 2022-05-08 by oliexdev