oliexdev/openScale #822: RGB (pure red and green) arrow colors do not have the best contrast

nekohayo, 2022-01-24:
This is a minor issue, but also fairly easy to fix. It seems like the red and green colors are juste the pure RGB values, and this makes them unpleasant to look at depending on which theme (light vs dark) you are using. Pure red is hard to see on a dark background, and pure green is hard to see on a white background.

It would probably be better to use the [Tango color palette](http://tango.freedesktop.org/Tango_Icon_Theme_Guidelines#Color_Palette) and use two sets of colors, for dark vs light background theme. Here are some ideas (just guessing which colors could be suitable, you'll have to try them out):

* "Red" arrow: dark red (`#a40000`) for light theme and orange (`#f57900`, or `#fcaf3e`?) for dark theme
* "Green" arrow: `#4e9a06` for use in the light theme, and `#8ae234` for dark theme

oliexdev, 2022-04-15:
If somebody is willing to send a PR I would be happy.

alt-shreya, 2023-01-28:
Has this been resolved? If not, I'd like to work on this.

oliexdev, 2023-01-28:
no go ahead :)

oliexdev, 2024-11-17:
Thanks for your idea but I think I will stick to the current colors as they look better with the current blue one.

closed 2024-11-17 by oliexdev