oliexdev/openScale #823: monthly reminder to weight

Micha-Btz, 2022-01-25:
**Is your feature request related to a problem? Please describe.**
The app offers a weekly reminder to weight. This span is to short for me.

**Describe the solution you'd like**
A reminder to get weight monthly.

**Additional context**
As workaround I use the weekly reminder and measure only every fourth.

Would be nice, TIA Micha

oliexdev, 2022-04-15:
In general a good idea but then you have to choose which day in the week you would get a reminder!? It getting complex then because of the different days per months and so on.

Micha-Btz, 2022-04-17:
I understand, what about to choose a regular  day of month. Maybe the 15th of the month or so. Not really a monthly thing because of different length off month but not really important if 30 or 31 or even 28 days.

oliexdev, 2022-04-23:
Maybe its not important for you but for others. If I implemented this way I surely get a complain why its not adjustable.

Micha-Btz, 2022-04-26:
I understand, It is to much effort.

closed 2022-04-26 by Micha-Btz