oliexdev/openScale #824: Import (very)  old database results in error

Micha-Btz, 2022-01-25:
Hi again,
I have found an old database export from openscale. ( Sadly I don't know how old).

I get an error while importing, missing header. I add a screenshort.
[image]

```
in ~ ✦ ❯ gocat -p com.health.openscale
         OpenGLRenderer  D  endAllActiveAnimators on 0xb4000071c9dcf700 (MenuPopupWindow$MenuDropDownListView) with handle 0xb4000070e9b441c0
  NetworkSecurityConfig  D  No Network Security Config specified, using platform default
                         D  No Network Security Config specified, using platform default
  CustomActivityOnCrash  I  CustomActivityOnCrash has been installed.
         ealth.openscal  W  Accessing hidden field Ljava/util/Collections$SynchronizedCollection;->mutex:Ljava/lang/Object; (greylist-max-o, reflection, denied)
                         W  Accessing hidden method Ljava/util/Collections$SynchronizedSet;-><init>(Ljava/util/Set;Ljava/lang/Object;)V (greylist-max-o, reflection, denied)
                         W  Accessing hidden method Ljava/util/Collections$SynchronizedCollection;-><init>(Ljava/util/Collection;Ljava/lang/Object;)V (greylist-max-o, reflection, denied)
           AdrenoGLES-0  I  QUALCOMM build : 4783c89, I46ff5fc46f
                         I  Build Date : 11/30/20
                         I  OpenGL ES Shader Compiler Version: EV031.31.04.01
                         I  Local Branch : QPR2
                         I  Remote Branch :
                         I  Remote Branch :
                         I  Reconstruct Branch :
                         I  Build Config : S P 10.0.4 AArch64
                         I  Driver Path : /vendor/lib64/egl/libGLESv2_adreno.so
                         I  PFP: 0x016dd090, ME: 0x00000000
            AdrenoUtils  W  <ReadGpuID_from_sysfs:197>: Failed to open /sys/class/kgsl/kgsl-3d0/gpu_model
                         W  <ReadGpuID:221>: Failed to read chip ID from gpu_model. Fallback to use the GSL path
         ealth.openscal  I  WaitForGcToComplete blocked RunEmptyCheckpoint on ProfileSaver for 6.573ms
         OpenGLRenderer  D  endAllActiveAnimators on 0xb4000071c9c5cd50 (MenuPopupWindow$MenuDropDownListView) with handle 0xb4000070e9b60c00
CompatibilityChangeRepo  D  Compat change id reported: 147798919; UID 10291; state: ENABLED
         OpenGLRenderer  D  endAllActiveAnimators on 0xb4000071c9b32bb0 (MenuPopupWindow$MenuDropDownListView) with handle 0xb4000070e9b5da50
                         D  endAllActiveAnimators on 0xb4000071c9dcffa0 (MenuPopupWindow$MenuDropDownListView) with handle 0xb4000070e9b417c0
```
Sadly the logcat seems to not include some helpfully informations. If you need additional info, just ask.

Pixel 5 redfin, LOS 18.1 without gapps.

TIA Micha

Blake-CB, 2022-03-31:
I get the same issue but the backup is not that old.  It is from early 2021.

oliexdev, 2022-04-15:
Did you import the database using settings->backup->import or using the three dots upper right?

Micha-Btz, 2022-04-17:
The three dot upper right, if I remember correct.

oliexdev, 2022-04-23:
Yeah then it can't work because the three dots in the upper right you can import/export CSV files not your backup files. Please use the settings->backup->import.

closed 2022-04-23 by oliexdev