oliexdev/openScale #830: I wanna buy a scale, but I can't figure out what would be the smoothest experience.

Kent-Erik, 2022-02-17:
Hello!

As the title says, I would like to buy a scale to use with the open scale app. I want to be able to see as much body metrics as possible. 
Visceral fat in particular is important to me. Initially I was thinking to get the Xiaomi Mi Body Composition Scale 2, but I can read that it used to be a problem that you couldn't monitor body metrics on that scale. Is this still the case? 

If the Xiaomi can't monitor other body metrics than just the weight, my second question is what scale that is the best option, in terms of smooth experience and body metrics compatibility?

Thanks in advance, and thank you for creating such an awesome project.

Neo2SHYAlien, 2022-04-09:
Hi I have the very same scale from few weeks and works fine with my Pixel 4a.

oliexdev, 2022-04-15:
Xiaomi scale v2 should work also with the body metric (but could be possible they have change something on the firmware that would break the openScale support)
Please read the wiki page for more information.

closed 2022-04-15 by oliexdev