oliexdev/openScale #837: Scales in Australia? Adding support for Target / Kmart scales?

chrisjdixon, 2022-03-06:
I'm in Australia looking for a scale suitable to use with this app but can't find any to buy among [the list of supported scales](https://github.com/oliexdev/openScale#supported-bluetooth-scales-rocket). Cursory research suggests [this Kmart Anko scale](https://www.kmart.com.au/product/bluetooth-body-analysis-scale/1585830), or [this Target generic scale](https://www.target.com.au/p/bluetooth-body-fat-electronic-scale/63886047) have all the hardware I'd want. They are likely sold under different brands overseas.

Looks like there aren't many great options down under and I've found [this old thread](https://github.com/oliexdev/openScale/issues/309) where @freman has had a crack at getting a Senssun Kmart scale from '18 working, and although they seemed to make all reasonable efforts it seems that scale still isn't supported. I'm nowhere near as good at programming as everyone here so I'm reluctant to buy a scale and attempt to make it work.

I don't have the time or mental capacity to start a new project of figuring all this out but I'd like to be able to just use one of these without a hassle and I want to support open source programs like this. Is there anyone here in Australia who knows how to add support for new scales who wouldn't mind doing so if I sent one of these to them?

Or does anyone know of a better option readily available in Australia?

chid, 2022-03-07:
not sure if this is helpful but this is available (I have one) - https://www.kogan.com/au/buy/xiaomi-body-composition-smart-scale-2-xiaomi/

Nine-H, 2022-03-11:
I got the Kmart scale. Me mum got it for me a couple of years ago. I can take a look at capturing the required log files over the weekend

chrisjdixon, 2022-03-11:
Thank you @chid, I've since ordered the Xiaomi Smart Scale 2 and it's arrived. Side note: Kogan has since spammed me with junk mail despite "unsubscribing" several times so I recommend avoiding them.

So far this Xiaomi scale seems ok but have weighed myself only once. It's listed my body fat at 30% which I hope is extremely inaccurate 😛.

I was about to order the popular Renpho ES-CS20M but saw [too many issues here](https://github.com/oliexdev/openScale/issues/696) discussing problems with it.

@Nine-H, although I went with the Xiaomi, you adding support for the Kmart scale would be a great thing to do and I'm sure people will come to benefit from such kindness.

Nine-H, 2022-03-11:
I took a look and the device name is IF_B2 so it's just a rebranded OEM Senssun Body Fat scale. The app reports that the device is not supported even after the initial PR was merged. Possibly the protocol changed or there's something else I need to do to pair it up properly to my smartphone (pls advise). I can capture a new bluetooth log if it's not user error.

chid, 2022-03-13:
> Thank you @chid, I've since ordered the Xiaomi Smart Scale 2 and it's arrived. Side note: Kogan has since spammed me with junk mail despite "unsubscribing" several times so I recommend avoiding them.

that's good to hear (and that is the unfortunate thing about dealing with Kogan but they seem to have products that aren't easily available otherwise in Australia), I actually don't use openScale so hopefully it works well for you. I found this repo as I was investigating how the implementation in the homeassistant module performed calculations, and trying to find a source for how the static values had been determined and it appears most of it was taken from decrypted/reversed Android app sources.

Nine-H, 2022-03-14:
I captured a log and I'm working on reversing it.

kyle6513, 2026-02-14:
Hi Team,

I've combined a bit of information on this here.
https://github.com/oliexdev/openScale/issues/1009
and
https://github.com/oliexdev/openScale/issues/309
and indeed:
https://github.com/zhangjing02/MySenssunSdk/tree/master/sdk/src/main/java/cn/senssun

Seem to point to a lot of useful information.
However I have the Kmart brand (likely) from around 5-6 years ago. No doubt the protocol has changed.

My scale is indeed IF_B2 and I've pulled some logs from the original firmware. I'm certain that there are likely some hidden commands that the scale doesn't use but it seems that there's a few specific packets you can send. Pretty much everything is little endian (hex value literally represents the number) so it's super easy/simple to read.

[User profile] - this can be used to setup your profile (user age, height, gender) all commands have a simple addition checksum at the end.
```
[USER PROFILE]	10 00 00 C5 13 03 01 00 00 02 00 A0 1F 02 00 00 41 1B
                                           |  |  |  |  |         |_|
                                           |  |  |  |  |         |
                                           |  |  |  |  |         |_Checksum (???) Still not 100% on this
                                           |  |  |  |  |
                                           |  |  |  |  |___?? Not sure what this is - need further dumps changing options
                                           |  |  |  |___Age (Years)
                                           |  |  |___Height (CM)
                                           |  |___Gender (female/male:0/1)
                                           |___User number                      
```                                                                                       
This MUST be called before you do anything; specifically you're basically syncing with the scales themselves and selects which profile you're going to be using to call the following history command. This will spit out each history value starting with the oldest to the newest as follows across two packets. All bigger returned data ends up across two packets:
```
Packet 1
Byte: | 00 01 02 03 | 04 05 06 | 07 | 08 | 09 | 10 11 | 12 13 | 14 15 | 16 17 | 18 19 |
Data: | 10 00 00 C5 | 1D 03 82 | 01 | 00 | 02 | 02 71 | 00 EC | 00 85 | 02 0D | 01 75 |
      |             |          |    |    |    |       |       |       |       |       |
      [---Header---] [---CMD---][ST] [PD] [US] [ WGT ][ FAT ] [ MUS ] [ HYD ] [ BONE]

Packet 2
Byte: | 20 21 | 22 23 | 24 25 26 27 | 28 |
Data: | 00 23 | 05 8F | 00 6C CD 6E | 6C |
      |       |       |             |     |
      [ VISC ] [ BMR ][  TIMESTAMP  ][ CS ]

[Header]: Standard header on all commands
[CMD] : Command number - 0x1D0382
[ST] \t: Can be 00 for live, any number for entry, and FF for the empty packet with reference time (more on that later)
[PD]: Padding as far as I can tell; haven't seen it change (maybe it does?!)
[US]: User number, 0x01-> 0x08, based on memory outputs
[WGT]: LE weight in Kilograms (62.5kg in this example) (I've not seen it return in pounds yet
[FAT]: fat percentage, with decimal IE 23.6% in this example.
[MUS]: Muscle index; not 100% sure how to get the actual %age yet. 133 here, maybe 13.3%? (it doesn't align with the app)
[HYD]: Hydration percent, 1 LSD 52.5% here.
[BONE]: Bone percentage, two LSDs: 3.73%
(END OF PACKET)
[VISC]: This is a guess since it doesn't align anywhere but Visceral muscle 35; so maybe 3.5%? UNSURE
[BMR]: This is in KJ; no decimals - 1423KJ in this example
[TIMESTAMP]: Offset in seconds from either last measurement or the baseline - not 100% certain yet.
[CS]: Checksum - Sum(0-36) + 0x17

The "time sync" history packet is similar to the above however the following changes:
[ST]: just FF
[PD]: Padding
[US]: User number
[WGT]: 0x00 always
[FAT]: 0x00 always
[MUS]: 0x00 always
[HYD]: 0x02AF always (Not sure why?)
[BONE]: 0x00 always
(END OF PACKET)
[VISC]: Total number of sessions/weights for this individual
[BMR]: Unknown - likely relating to the sessions/weights. could also be a bitmask; needs more diverse samples.
[TIMESTAMP]: Baseline number. What the scale currently _thinks_ the time is. If you request history again the next FF entry will have a seconds offset equivalent to the time passed. It seems to be counting up from around 11 August 1990 for whatever reason; likely a certain date +/- leap years (it doesn't care about years, just duration between syncing).
[... 25 lines cut ...]

[RCV]		10		00		00		C5		12		03		87		08		04		00		00		00		00		00		00		00		00		A8

[RCV]		10		00		00		C5		12		03		87		08		05		00		00		00		00		00		00		00		00		A9

[RCV]		10		00		00		C5		12		03		87		08		06		00		00		00		00		00		00		00		00		AA

[RCV]		10		00		00		C5		12		03		87		08		07		00		00		00		00		00		00		00		00		AB

```
To be clear my model on the base of my scales states Mode: 42489146 with a manufacture date of 28/20 (28th week of 2020).

There's also the live data packet; that streams data but once the fat check is done it will give you two packets including all the same information as history; including pounds for some reason. I'm not sure if there's a switch somewhere to "switch" all of your history to pounds and I haven't attempted the physical switch as yet - more work to be done.

Nevertheless the above should get us 90% of the way there. Keep in mind that once you read history that gets 'popped' off the device and potentially lost if you don't capture it.

[... cut ...] the python code online; but at the moment we're working!

[image]

[image]

You need to ensure you initialize the scale with a profile before you're able to complete a "weight" in some instances. It's a bit fussy when it comes to performing the fat logging. Especially if your "last weight" is greater than +/- 3kg or so (guess, not tested).

If anyone else has a scale and an android phone I can walk you through getting me some more sample data for your scale to see if it has the same information.

kyle6513, 2026-02-14:
annnd here's a super rough branch that manages to capture all the data when a successful fat scan is completed.

https://github.com/kyle6513/openScale/tree/add-kmart-IFB2-scale

I'll add the history stuff over the coming week/month and then we can get this merged for y'all.