oliexdev/openScale #838: Table View: Show day name with date

jmorgannz, 2022-03-10:
**Is your feature request related to a problem? Please describe.**
Cognitively difficult to look up weight on a particular *day*  
Requires secondary lookup to calendar to find day from date.  

**Describe the solution you'd like**
Show (optionally?) day name in DDD format as part of date format in table view date column.  
Additionally, space can be reclaimed by (optionally) showing date time format in HHmm format.  

**Describe alternatives you've considered**
None  

**Additional context**
None

oliexdev, 2022-04-15:
It's no problem to show the name of the day but the space is a problem. Add an option to enable/disable certain columns would require much effort in the implementation.

jmorgannz, 2022-04-15:
There is actually heaps of space because each row is double lined - the date part already wraps to fit the year into it.  
You could put the day after the year.  
If you were willing to be a bit less 'pure', you could also put the day on the second line of the time column :S  

See example on top two entries.  
https://imgur.com/a/w5YMJiZ

oliexdev, 2022-04-15:
Seems not so easy, every language has their own time/date format with less and more space used. Currently I use the MEDIUM DateFormat by Android, I could use FULL but then everything is written out. Sure I could use a custom format but then it won't be correct anymore in a certain language.
I could change the FULL format on the top in the measurement view if you click on a item?

jmorgannz, 2022-04-15:
Ahh, dang I thought it might be regionalised.  
It's difficult because IMO from a UX point of view, the day name is a lot more meaningful to a human than just the numbers are.   
The date becomes kind of useless and you have to do a second lookup to a calendar to figure out when things happened.  
Whereas if there are day names, a lot of the time a human will now 'Oh yes that was 2 Wednesdays ago' type of thing.

oliexdev, 2022-04-23:
In the DateMeasurementView.java the format will be set.

oliexdev, 2023-01-29:
@jmorgannz I have been refactored the overview and table layout, now the day name is also displayed. Please try out the latest dev-build.

closed 2023-01-29 by oliexdev

jmorgannz, 2025-05-07:
Hey @oliexdev 
I know this was done ages ago but just wanted to say thanks for adding the day name in the table view. I have been using it with that for ages now and it is much more enjoyable and efficient