oliexdev/openScale #842: Import 'Sanitas export data'

kbot82, 2022-03-25:
Since the Sanitas app never connected to the scale with my Xiaomi phones, I used my old iPhone 5s, but soon it will break down. 
That's why I want to switch to open scale and want to import the previous scale data of the last years. 

But the Sanitas app only exports the data in pdf format and open scale says "header line not valid: %PDF 1.4". 
Would be nice, if I could import this data into open scale.

This is the export file of Sanitas:
[Sanitas Export File](https://drive.google.com/file/d/1ZIUMeJid8E9LY4jSyBRzTDAF8MUF0iyj/view?usp=drivesdk)

oliexdev, 2022-04-15:
Sorry openScale doesn't support an import of a PDF file. You can try with a 3rd party tool to generate from the PDF file a csv file.

closed 2022-04-15 by oliexdev

claus-hultschig, 2024-04-21:
Thanks for providing open scale.
I did run into the same issue as @kbot82 when try to migrate away from the vendors app..
I could convert the silly pdf created by the "healtcoach" app of sanitas using tabula (https://tabula.technology/).
So far I failed in inporting the table I did obtain. 
Where  can I find an example of the colums needed for importing into open scale? I failed to find anything in the code.