oliexdev/openScale #848: Can't Detect FitIndex ES-M26

Plokmju22, 2022-04-22:
**Describe the bug**
I'm unable to detect my FitIndex ES-M26 using the app. The scale shows up fine in the FitIndex app so I think the scale is working, but it doesn't show up at all in openScale. I'm confused because it's listed as supported.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on **Settings**
2. Click on **Bluetooth**
3. Click on **Your Bluetooth scale - **
4. Step on the Scale
5. Scale isn't detected

Reproduced with [a7c4d54](https://github.com/oliexdev/openScale/commit/a7c4d542529136eafa5549d1111ca00a4e0be750)

**Expected behavior**
Scale appears when searching and can be paired with openScale.

oliexdev, 2022-04-23:
see #752 but I will remove the scale from the supported list.
Could you post your scale max address?

closed 2022-04-23 by oliexdev

Plokmju22, 2022-04-23:
Yep!

MAC Address is ED:67:38:F4:F2:33

I will review #752 right now; I appreciate all of your help.