oliexdev/openScale #849: Cannot get data from Xiami Scale v2

Strubbl, 2022-04-22:
**Describe the bug**
I get no data although the connection to scale was established

**To Reproduce**
Steps to reproduce the behavior:
1. install dev build (commit a7c4d542529136eafa5549d1111ca00a4e0be750)
2. Start app
3. add user
4. select scale
5. finish starting wizard
6. enable bluetooth via button
7. wait
8. see toast with connecting state
9. see toast with connected state
10. wait
11. see toast with disconnected state
12. see no data has arrived

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): yes

**Expected behavior**
Downloaded data from the scale can be seen e.g. in the table view of the app

**Additional context**
Device: Samsung SM-S901B
OS version: Android 12 (SDK 31)

**Debug log**
see https://gist.github.com/Strubbl/f04909c7d3be78318b95967eebd1f35c

oliexdev, 2022-04-23:
Did you also don't get measurement when you weight yourself?

Strubbl, 2022-04-23:
No, i can confirm i can get new measurements. So only the old ones are missing. I can import them via csv. I think this issue can be closed.

closed 2022-04-23 by Strubbl