oliexdev/openScale #853: Wyze Scale (WHSCL1) Support

HerbFargus, 2022-05-06:
**Wyze Scale (WHSCL1)**

Vendor App:
https://play.google.com/store/apps/details?id=com.hualai&hl=en_US&gl=US

Product Page
https://www.wyze.com/products/wyze-scale

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**
_Attach 3 log files with the corresponding true values, read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#1-acquiring-some-bluetooth-traffic) for further information._

1. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male, body height: 6'4" ft, Year of Birth 1991
```
measured true values in the vendors app for the 1. HCI Snoop log file:

| Number         |                   1 |
|----------------|--------------------:|
| Date and Time  | 2022.05.05 09:27 PM |
| Weight         | 168.4lb             |
| BMI            | 20.5                |
| Body Fat       | 15.1%               |
| Muscle Mass    | 134.2lb             |
| Body Water     | 62.2%               |
| Lean Body Mass | 143.0lb             |
| Bone Mass      | 8.8lb               |
| Protein        | 17.5%               |
| Visceral Fat   | 7                   |
| BMR            | 1771                |
| Metabolic Age  | 27                  |

Log 1:
[btsnoop_hci_168.4.log](https://github.com/oliexdev/openScale/files/8637630/btsnoop_hci_168.4.log)

2. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male, body height: 6'4" ft, Year of Birth 1991
```
measured true values in the vendors app for the 2. HCI Snoop log file:

| Number         |                   2 |
|----------------|--------------------:|
| Date and Time  | 2022.05.05 09:31 PM |
| Weight         | 188.7lb             |
| BMI            | 23.0                |
| Body Fat       | 21.4%               |
| Muscle Mass    | 139.3lb             |
| Body Water     | 57.6%               |
| Lean Body Mass | 148.3lb             |
| Bone Mass      | 9.0lb               |
| Protein        | 16.2%               |
| Visceral Fat   | 8                   |
| BMR            | 1823                |
| Metabolic Age  | 27                  |

Log 2:
[btsnoop_hci_188.7.log](https://github.com/oliexdev/openScale/files/8637634/btsnoop_hci_188.7.log)

3. Bluetooth HCI Snoop log file
user settings in the vendors app:
```
sex male, body height: 6'4" ft, Year of Birth 1991
```
measured true values in the vendors app for the 3. HCI Snoop log file:

| Number         |                   3|
|----------------|--------------------:|
| Date and Time  | 2022.05.05 09:35 PM |
| Weight         | 209.2lb             |
| BMI            | 25.5                |
| Body Fat       | 29.5%               |
| Muscle Mass    | 138.4lb             |
| Body Water     | 51.6%               |
| Lean Body Mass | 147.5lb             |
| Bone Mass      | 9.0lb               |
| Protein        | 14.5%               |
| Visceral Fat   | 9                   |
| BMR            | 1815                |
| Metabolic Age  | 29                  |

Log 3:
[btsnoop_hci_209.2.log](https://github.com/oliexdev/openScale/files/8637637/btsnoop_hci_209.2.log)

**Step 3: Discover Bluetooth services and characteristic**
_Read [here](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic) how to create the openScale debug file_

OpenScale Log:
[openScale_2022-05-05_13-39[1].txt](https://github.com/oliexdev/openScale/files/8637640/openScale_2022-05-05_13-39.1.txt)

BLE Scanner:
[WHSCL1DeviceInfo.txt](https://github.com/oliexdev/openScale/files/8637641/WHSCL1DeviceInfo.txt)

Wyze API example (from a separate project, may not be relevant for this but found it interesting)
https://github.com/JoshuaMulliken/ha-wyzeapi/issues/93#issuecomment-970230075
[ScaleVariables.txt](https://github.com/oliexdev/openScale/files/8637660/ScaleVariables.txt)

karanrajpal14, 2023-02-20:
So I'm trying to fetch the logs for the [Wyze Scale X](https://www.wyze.com/products/wyze-scale-x), but I wonder how you made sense of the data. Opening the `btsnoop_hci.log` in notepad results in gibberish and when I open it up in wireshark, I do see my phone and my device listed there, but I can't find the packets in which my weight or other details are listed. Mind helping me out?

HerbFargus, 2023-02-20:
As the data in the logs is often obfuscated or in byte code I used the wyze app to pull the values associated with the logs eg I started the logging, weighed myself, noted the values in the wyze app, stopped the logging, saved the log, rinse and repeat etc. In theory logical data should be able to be derived from a comparison of the resulting tables and the logs but I didn't get much further than gathering the data as I moved on to other projects. I left the logs here as a basepoint for others to compare against. As a note I have the original wyze scale, don't know how much the API or Bluetooth logs differ between this and the X.

famewolf, 2023-12-27:
Has adding this particular Hardware been abandoned?

JesusFreke, 2024-02-03:
here's a hand-annotated protocol dump for you :)

```

-------------------------------------------------------------------------------
Send public key to scale
> 00f000089ad9fd9600000000
(public key = 0x96FDD99A)

-------------------------------------------------------------------------------

received public key from scale
< 42f000089cebc88600000000
(public key = 0x86C8EB9C)

-------------------------------------------------------------------------------

send CMD_SYNC_TIME
(raw)
> [22, 0, 7, 0, 1, -88, -85, -58, -68, 101, 1]
(rawhex)
> 16 00 07 00 01 A8 AB C6 BC 65 01
(encrypted)
> 1101000bd2d3e619fd6e58d6dfb23437dbc5486a

time = 0x65BCC6AB = time in seconds since epoch = Friday, February 2, 2024 10:40:43 AM GMT
os = 0x01  (probably specifies the type of timestamp? 0x01 being unix timestamps?)

-------------------------------------------------------------------------------

receive sync type reply
(raw)
< 53010007aff48a08e48a47ff
(unencrypted)
< 2201030001a800

last byte = 00 = success

-------------------------------------------------------------------------------

send CMD_USER_LIST_NEW

(raw)
> [22, 0, 2, 0, 13, -88]
(encrypted)
> 1201000668e95d0a577b99ff

-------------------------------------------------------------------------------

uninteresting empty CMD_CUR_WEIGHT_DATA message

(raw)
< 54010033b21b07b51f9eec6bea5c20685e15c628ea5c20685e15c628ea5c20685e15c628ea5c20685e15c628ea5c20685e15c628e4a12a355257a7aa
(unencrypted)
< 22012f0008a864000000000000000000000000000000000000000000000000000000000000000000000000000000000000c300

-------------------------------------------------------------------------------

CMD_USER_LIST_NEW response

(raw)
< 55010020542e23a5086cd3aa4fd800f6c5846ad1e40f592000525f04ed8cc484d0ddd71a
(unencrypted)
< 22011c000da801 2f319e4a8a8db394f3fa7ef7f9fdaa7f a618 01 2c b4 01 00 440c

user id = 2f319e4a8a8db394f3fa7ef7f
weight = 18A6 = 6310/100 = 63.1kg
sex = 1 = M
age = 2c = 44
height = b4 = 180cm = ~5'11"
[... 590 lines cut ...]

        reply_class = REPLY_MAP.get(decrypted_buf[4])
        if not reply_class:
            print("Received unexpected message type: 0x%02x" % decrypted_buf[4])
            return

        self._read_queue.put(reply_class(decrypted_buf[0:payload_length]))
```

```
def send_message(self, buf):
        print("sending message: %s" % binascii.hexlify(buf))
        message = io.BytesIO()
        message.write(struct.pack("4b", self._get_next_frame() + 0x10, 1, 0, len(buf)))

        padded_payload = io.BytesIO()
        padded_payload.write(buf)
        while len(padded_payload.getvalue()) % 8 > 0:
            padded_payload.write(b'\x00')

        for i in range(0, int(len(padded_payload.getvalue()) / 8)):
            message.write(xxtea.encrypt(padded_payload.getvalue()[i*8:(i+1)*8], self._xxtea_key,  padding=False))

        self._write_queue.put(message.getvalue())
```

JesusFreke, 2024-02-03:
I forgot to mention, this is from the original wyze scale, not a wyze scale x.

JesusFreke, 2024-02-08:
I've published some proof-of-concept (python) code for talking to the scale:  https://github.com/JesusFreke/wyze_scale_tool

There should be enough info there to be able to implement the protocol in java for android

HerbFargus, 2026-10-20:
used Claude to help finish the testing i started with the scale, and got it working on the app with one caveat of needing a one time wyze key, addressed in PR https://github.com/oliexdev/openScale/pull/1569

JesusFreke, 2026-10-21:
hmmm. I don't recall needed to get any key when I was writing my tool. It's been a while though, I may be misremembering.

HerbFargus, 2026-10-21:
It's only for the first model, it shouldn't be necessary for the later models

famewolf, 2026-10-28:
Can someone point me to the actual steps to ADD this scale in openscale? I have the device id needed for the WHSCL1.