oliexdev/openScale #856: Crash on start

SzaboAdam72, 2022-05-10:
Hello!

Since the last update, the application has crashed on startup.

I included the LOG.

Please Help!

Thanks!

Adam

Build version: 0.3.3 
Build date: 1981-01-01 01:01:02 
Current date: 2022-05-10 06:23:05 
Device: Xiaomi M2002J9G 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.IllegalArgumentException: com.health.openscale.sync: Targeting S+ (version 31 and above) requires that one of FLAG_IMMUTABLE or FLAG_MUTABLE be specified when creating a PendingIntent.
Strongly consider using FLAG_IMMUTABLE, only use FLAG_MUTABLE if some functionality depends on the PendingIntent being mutable, e.g. if it needs to be used with inline replies or bubbles.
at android.app.PendingIntent.checkFlags(PendingIntent.java:375)
at android.app.PendingIntent.getBroadcastAsUser(PendingIntent.java:645)
at android.app.PendingIntent.getBroadcast(PendingIntent.java:632)
at t4.b.e(:3)
at t4.b.p(:6)
at t4.d.run(Unknown Source:52)
at java.lang.Thread.run(Thread.java:920)

oliexdev, 2022-05-15:
Which service you want to sync for? GoogleFit? I can't reproduce this error? When does this error occure? When you openScale sync or openScale? What did you?

oliexdev, 2022-05-15:
Ok found out that the MQTT service library (org.eclipse.paho:org.eclipse.paho.android.service:1.1.1) is not compatible with Android 12.

oliexdev, 2022-05-15:
just released version 0.3.4 of openScale sync that should fix this error.

closed 2022-05-15 by oliexdev

SzaboAdam72, 2022-05-17:
Thank you!

 

It’s fine!

 

Ádám