oliexdev/openScale #858: iHealth HS3 not shown in device list

mkre, 2022-05-11:
Hi,

I want to pair my iHealth HS3 scale, but I cannot see it in the in-app bluetooth device list. I can pair it in the Android settings menu, though. One thing which caught my eye is that the source code refers to the scale as "iHealth HS33FA4A" [here](https://github.com/oliexdev/openScale/blob/master/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothIhealthHS3.java#L55=) but in my Android bluetooth device list it is called "iHealth HS37B017". Not sure if that's a problem, though.

Thanks for your help,
Moritz

oliexdev, 2022-05-29:
please unpair your Bluetooth scale on your Android device and try again to search for it.

mkre, 2022-05-30:
Thanks for the reply. I have sent back the scale in the meantime and I'm currently looking for a better supported device without too many (useless for me) bells and whistles like body fat measurement etc.. Any recommendations are welcome. Also, feel free to close this ticket.

closed 2022-06-04 by oliexdev