oliexdev/openScale #860: Xiaomi Mi Smart Scale 2 - read only the last measurement

Yenya, 2022-05-16:
**Is your feature request related to a problem? Please describe.**
I bought a scale named "Xiaomi Mi Smart Scale 2" - it supports only weight measurement and nothing else - but it apparently stores all the measurements. So when I am not the sole user of this scale, every time I read measurements to OpenScale, I get also the measurements by my family members. Which I not only don't want to be imported to OpenScale, I don't want to see them at all (and the family members don't want me to see them either :-)

Fortunately for me, my own weight is sufficiently different to that of my family members, so I can filter out mine measurements. But the rest of my family is not so lucky :-)

**Describe the solution you'd like**
Maybe OpenScale can support "read only the last measurement" mode instead of "download everything.

**Describe alternatives you've considered**
Explore whether the Mi Smart Scale 2 supports deleting the measurements from the device. If it does, another option would be the "download and delete" mode of OpenScale.

Thanks for considering this.

-Yenya

vonguyenkhang, 2022-05-28:
There's already support for multiple users in the app. Create new user for each family member, key in their initial weights, and the app can auto differentiate the measurements of the users. 
Measurement deletion works for me.
I'm using Mi Scale (v1)

oliexdev, 2022-05-29:
Enable also the smart user assignment in the Bluetooth settings after you added multiply users on openScale.

closed 2022-05-29 by oliexdev