oliexdev/openScale #865: Beurer BF 980

jokay, 2022-06-16:
**Scale name**

[Beurer BF 980](https://www.beurer.com/web/gb/products/wellbeing/weight-and-diagnosis/diagnostic-bathroom-scale/bf-980.php)

**Step 1: Read the general reverse engineer process**

- [x] I have read the [how to reverse engineer a Bluetooth 4.x wiki](https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale)  entry

**Step 2: Acquiring some Bluetooth traffic**

I cannot provide them as my decision on buying this scale depends on the support by openScale :wink:

jokay, 2023-01-04:
[beurer HealthManager Pro](https://play.google.com/store/apps/details?id=com.beurer.healthmanager) can be used without account and without internet.

Therefore no additional value from this app, issue closed.

closed 2023-01-04 by jokay

2D-Flake, 2023-09-07:
The [app](https://play.google.com/store/apps/details?id=com.beurer.healthmanager) runs without an online account for most scales, but to add the scale BF 980 a online account is needed.

2D-Flake, 2023-09-07:
@jokay Thank you for replaying. A price I would gradly pay for privacy;)
When I try to add the BF980 this pops up: 
[image]

Without wifi the scale and the app do not sync, deleting the online account removes the device pairing. How did you manage to pair the BF980 without an online account? Getting to where you are would be awesome!

jokay, 2023-09-07:
No idea, may be the existing account(s) were not affected or they may have indeed changed the workflow.

As I no longer use the scale, I can't say for sure :wink: