oliexdev/openScale #866: Application crashes when connecting to BF720

bertaga, 2022-06-19:
**Describe the bug**
When connecting to the BF720 scale for the first time the application crashes if there is no user yet and crashes after showing the list of users if there are some.

**To Reproduce**
Steps to reproduce the behavior:
1. Clean install the app
2. Add the BF720 scale
3. Connect to it
4. The app crashes after showing the users list if there are

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): yes

**Expected behavior**
Being able to select a user on the scale or create one

**Additional context**
I was able to use it on a previous phone but it was also crashing a lot on the first connect so it was kind of magic work to make it connect.
It also seems like each time I retry to connect after cleaning the app storage it creates a new user on the scale with default values.

**Debug log**
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-06-19 09:43:47 
Device: Google Pixel 5 
OS version: Android 12 (SDK 32) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
 at java.util.Objects.requireNonNull(Objects.java:245)
 at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
 at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeTargetWeight(BluetoothBeurerBF105.java:132)
 at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeUserDataToScale(BluetoothBeurerBF105.java:72)
 at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep(BluetoothStandardWeightProfile.java:196)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
 at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
 at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.handleUserControlPointNotify(BluetoothStandardWeightProfile.java:290)
 at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onBluetoothNotify(BluetoothStandardWeightProfile.java:256)
 at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.onBluetoothNotify(BluetoothBeurerBF105.java:82)
 at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:482)
 at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:277)
 at android.os.Handler.handleCallback(Handler.java:938)
 at android.os.Handler.dispatchMessage(Handler.java:99)
 at android.os.Looper.loopOnce(Looper.java:201)
 at android.os.Looper.loop(Looper.java:288)
 at android.app.ActivityThread.main(ActivityThread.java:7876)
 at java.lang.reflect.Method.invoke(Native Method)
 at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1019)
```

gazmunro, 2022-06-21:
I'm able to reproduce this issue when pairing a BF720 for the first time.

`
Build version: 2.4.6
Build date: 1981-01-01 01:01:02
Current date: 2022-06-21 19:44:23
Device: Motorola moto g(50)
OS version: Android 12 (SDK 31)

Stack trace:
java.lang.NullPointerException: no valid characteristic provided
at java.util.Objects.requireNonNull([Objects.java:245](http://objects.java:245/))
at com.welie.blessed.BluetoothPeripheral.writeCharacteristic([BluetoothPeripheral.java:1202](http://bluetoothperipheral.java:1202/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothCommunication.writeBytes([BluetoothCommunication.java:303](http://bluetoothcommunication.java:303/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothCommunication.writeBytes([BluetoothCommunication.java:291](http://bluetoothcommunication.java:291/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothBeurerBF105.writeTargetWeight([BluetoothBeurerBF105.java:132](http://bluetoothbeurerbf105.java:132/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothBeurerBF105.writeUserDataToScale([BluetoothBeurerBF105.java:72](http://bluetoothbeurerbf105.java:72/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep([BluetoothStandardWeightProfile.java:196](http://bluetoothstandardweightprofile.java:196/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothCommunication.nextMachineStep([BluetoothCommunication.java:599](http://bluetoothcommunication.java:599/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothCommunication.resumeMachineState([BluetoothCommunication.java:210](http://bluetoothcommunication.java:210/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothStandardWeightProfile.handleUserControlPointNotify([BluetoothStandardWeightProfile.java:290](http://bluetoothstandardweightprofile.java:290/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothStandardWeightProfile.onBluetoothNotify([BluetoothStandardWeightProfile.java:256](http://bluetoothstandardweightprofile.java:256/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothBeurerBF105.onBluetoothNotify([BluetoothBeurerBF105.java:82](http://bluetoothbeurerbf105.java:82/))
at [com.health](http://com.health/).openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate([BluetoothCommunication.java:482](http://bluetoothcommunication.java:482/))
at com.welie.blessed.BluetoothPeripheral$1$[5.run](http://5.run/)([BluetoothPeripheral.java:277](http://bluetoothperipheral.java:277/))
at android.os.Handler.handleCallback([Handler.java:978](http://handler.java:978/))
at android.os.Handler.dispatchMessage([Handler.java:104](http://handler.java:104/))
at android.os.Looper.loopOnce([Looper.java:238](http://looper.java:238/))
at android.os.Looper.loop([Looper.java:355](http://looper.java:355/))
at [android.app](http://android.app/).ActivityThread.main([ActivityThread.java:8080](http://activitythread.java:8080/))
at java.lang.reflect.Method.invoke(Native Method)
at [com.android](http://com.android/).internal.os.RuntimeInit$[MethodAndArgsCaller.run](http://methodandargscaller.run/)([RuntimeInit.java:548](http://runtimeinit.java:548/))
at [com.android](http://com.android/).internal.os.ZygoteInit.main([ZygoteInit.java:1026](http://zygoteinit.java:1026/))
`

FreakyPancake, 2023-01-25:
I have the exact same problem. To make sure everything with the scale itself works properly, I have installed the original companion App from Beurer ("beurer HealthManager Pro") and there it all works as expected.
I tested the openScale App on various devices, using a clean install on each device via F-Droid and made sure to reset the scale before trying out the next device. The procedure I did was always the same:
1. Insert batteries into freshly reset BF720
2. Go through setup process with device bluetooth turned on before opening the app
3. On Overview Screen of openScale press on bluetooth symbol to pair app with scale.
4. Perform a measurement.

All 5 devices I used to test openScale crashed either during the setup process after I confirmed pairing the scale via the system pop-up or if was able to go through the setup without the app crashing, it crashed when I pressed the bluetooth symbol on the overview screen. 
Then after pressing on "restart app", all devices besides the Tab S6 lite crashed again when trying to reconnect to the scale by pressing the bluetooth symbol. On the Tab S6 lite, after restarting the app, it asked for user 1 pin and then transferred measurements as expected. Only the user that was created in openScale didn't sync to the scale like the beurer app does. Instead I had to use the default user 1 which has been on the scale. Since I'm not sure if the user that has been created inside openScale is supposed to be synced to the scale I can't say if that is expected behaviour or not.
I used the following devices: CAT S52, OnePlus One, OnePlus 6, Samsung Tab S6 lite and LG Velvet 4G.
With my LG I seem to only have saved one of the 2 crash reports. I suppose it was the report from the crash during setup but I can't say for sure.

Since I'm not able to provide any fixes myself I'm happy to do more testing as needed and provide the results. I hope this can be fixed since I wouldn't want to rely on the companion app provided by Beurer. 

CAT S52 setup crash:
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-24 18:18:42 
Device: Cat S52 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.RuntimeException: Unable to resume activity {com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4233)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4265)
	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:52)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2041)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7386)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:510)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
Caused by: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:878)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3080)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.FragmentController.dispatchResume(FragmentController.java:284)
	at androidx.fragment.app.FragmentActivity.onResumeFragments(FragmentActivity.java:434)
	at androidx.fragment.app.FragmentActivity.onPostResume(FragmentActivity.java:423)
[... cut ...]
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:277)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:8005)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1039)

```

FreakyPancake, 2023-01-25:
I also have a problem when connecting to the BF720. In my case it crashes either during the setup process after confirming pairing of the scale via the system pop-up or after pressing the bluetooth icon on the overview screen right after the setup process. The crash recurrs (on all devices besides my Samsung Tab S6 lite) after restarting the app after the first crash and pressing the bluetooth icon again. The above mentioned user list either doesn't show up at all or only shows up for the fraction of a second before the app crashes.
To make sure everything with the scale itself works properly, I have installed the original companion App from Beurer ("beurer HealthManager Pro") and there it all works as expected.
I tested the openScale App on various devices, using a clean install on each device via F-Droid and made sure to reset the scale before trying out the next device. The procedure I did was always the same:

1. Insert batteries into freshly reset BF720
2. Go through setup process with device bluetooth turned on before opening the app
3. On Overview Screen of openScale press on bluetooth symbol to pair app with scale.
4. Perform a measurement.

All 5 devices I used to test openScale crashed either during the setup process after I confirmed pairing the scale via the system pop-up or if was able to go through the setup without the app crashing, it crashed when I pressed the bluetooth symbol on the overview screen. 
Then after pressing on "restart app", all devices besides the Tab S6 lite crashed again when trying to reconnect to the scale by pressing the bluetooth symbol. On the Tab S6 lite, after restarting the app, it asked for user 1 pin and then transferred measurements as expected. Only the user that was created in openScale didn't sync to the scale like the beurer app does. Instead I had to use the default user 1 which has been on the scale. Since I'm not sure if the user that has been created inside openScale is supposed to be synced to the scale I can't say if that is expected behaviour or not.
I used the following devices: CAT S52, OnePlus One, OnePlus 6, Samsung Tab S6 lite and LG Velvet 4G.
With my LG I seem to only have saved one of the 2 crash reports. I suppose it was the report from the crash during setup but I can't say for sure.

Since I'm not able to provide any fixes myself I'm happy to do more testing as needed and provide the results. I hope this can be fixed since I wouldn't want to rely on the companion app provided by Beurer. 

CAT S52 setup crash:
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-24 18:18:42 
Device: Cat S52 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.RuntimeException: Unable to resume activity {com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4233)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4265)
	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:52)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2041)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7386)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:510)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
Caused by: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:878)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3080)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
[... cut ...]
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:277)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:8005)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1039)

```

oliexdev, 2023-01-25:
@FreakyPancake try out the latest developer build  https://github.com/oliexdev/openScale/releases/tag/dev-build

FreakyPancake, 2023-01-26:
@oliexdev Thanks for your reply. I tested the dev-build, but only used my daily driver, the LG Velvet 4G, to check if the problem was resolved. 

During the setup process I ran into another Issue. When pressing the button to add the scale nothing happened. I then went into Settings -> Bluetooth -> Your Bluetooth scale. Pressing on this launched a Pop-up "Grant location access in the Android Settings to search for Bluetooth devices. Optionally revoke it afterwards.". After pressing OK the App crashed with the following Error details:

```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-25 22:54:02 
Device: LGE LM-G910 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: Attempt to invoke virtual method 'void androidx.fragment.app.FragmentActivity.startActivity(android.content.Intent)' on a null object reference
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment$1.onClick(BluetoothSettingsFragment.java:150)
	at com.android.internal.app.AlertController$ButtonHandler.handleMessage(AlertController.java:175)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:8005)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1039)

```

This error was reproducible after restarting the App and produced the same error message again. Fresh install and manually granting App permissions for location access and nearby devices still gave the same error.

Fresh install and no manually granted permissions.
I tried pressing on the Bluetooth icon on the Overview screen, the same Pop-up showed up, but pressing OK here redirected me to Android location service settings instead of crashing. Activating GPS geolocation removed the Pop-up in both places (settings & Bluetooth icon) and therefore no crash. Also adding the scale during the setup process worked again. 
Having location turned off creates the Pop-up that redirects to the right place via the Bluetooth icon but somehow causes a crash via Settings -> "Your Bluetooth scale" and also prevents anything from happening when pressing "Add Bluetooth scale" during setup process.

Sadly after managing to get that to work and being able to add the scale during the setup process, the same as with the F-Droid version happened. Adding the scale during setup works fine, pressing on the Overview Bluetooth icon to connect to the scale generates the system Pop-Up "Pair to BF720?" but after Pressing Pair the App crashes like before with the following Error details:

```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2023-01-25 23:15:33 
Device: LGE LM-G910 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1229)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeTargetWeight(BluetoothBeurerBF105.java:132)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.writeUserDataToScale(BluetoothBeurerBF105.java:72)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onNextStep(BluetoothStandardWeightProfile.java:196)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.handleUserControlPointNotify(BluetoothStandardWeightProfile.java:290)
	at com.health.openscale.core.bluetooth.BluetoothStandardWeightProfile.onBluetoothNotify(BluetoothStandardWeightProfile.java:256)
	at com.health.openscale.core.bluetooth.BluetoothBeurerBF105.onBluetoothNotify(BluetoothBeurerBF105.java:82)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onCharacteristicUpdate(BluetoothCommunication.java:493)
	at com.welie.blessed.BluetoothPeripheral$1$5.run(BluetoothPeripheral.java:287)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:8005)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1039)

```

oliexdev, 2023-01-28:
@FreakyPancake for the first problem it should be fixed now with the latest dev version please try it out.

The second problem is it seems you have a slightly different BF720 model than @schweigel who did the PR #808 
It seems the Bluetooth charisteristic `0x0003` is not available on your model. You can check it if you do the step 2 in https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale

FreakyPancake, 2023-01-28:
@oliexdev Ok, so the first problem is fixed and the pairing and permission redirection works fine. 

I'm going to take a look at step 2 now, to see if there is any difference and will then add the logs here so you can take a look at them. Do you think that this can be resolved somehow so that the BF720 model I have can be used with openScale sometime in the future? Otherwise I would return it, but I only have till 31st of January to do so, that's why I'm asking.
I'm not sure if the revision/update of a scale model is noticeable somehow via the Art.-No. or some other number on the backside of the scale, but if that's the case then it could make sense to add that number to the supported scales list, so it can be easily checked if the scale is actually the version that's supported. What do you think?

schweigel, 2023-01-28:
Hi,

the photo shows the data of my BF720. I hope this will help.

[image]

oliexdev, 2023-01-29:
@schweigel thanks for your photo
@FreakyPancake what's is your Art-No. of the scale?

I cannot guarantee if we get your scale to work. You could try to comment out the method `writeTargetWeight();` Maybe it isn't needed for your scale?

FreakyPancake, 2023-01-29:
@schweigel thanks for the picture. 
Apparently my scale has the same Art.-No. In the lower right corner mine says "H 04" instead of "F 31" but that could mean anything e.g. production machine, production batch,... 
It was worth a try.
@oliexdev Do you mean from the code of your app and then compile it with the method commented out? Sorry I'm completely new to all of this. And when I did step 2 of reverse engineering I was expecting that the debug log would give me some kind of list of supported characteristics for my scale like ```0x0003``` for example. Does the log I received look like expected? What would I have to do to find out which inputs my scale is expecting? I was hoping that it all would work kinda out-of-the-box, but I'd be happy to learn how to add support for my scale model.
Thanks for help!
[image]

oliexdev, 2023-01-30:
@FreakyPancake yes that is what I mean, sure you need a little bit of Android programming experience.
You could also do it with the BLE Scanner APP see Step 2 Nr 6