oliexdev/openScale #869: Crash with xiaomi scale 2

esalaud, 2022-07-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): ???

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-07-17 09:06:19 
Device: Xiaomi Redmi Note 9 Pro 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:878)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:771)
	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:690)
	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:855)
	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:154)
	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:123)
	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:306)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1853)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1764)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1701)
	at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:488)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)

oliexdev, 2022-08-20:
Please give the permission to the location (GPS).

closed 2022-08-20 by oliexdev

esalaud, 2022-08-22:
Thann you.
But why localisation permission ?