oliexdev/openScale #872: Transparent widgets

basings, 2022-07-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

It's just a minor (subjective) enhancement. My home screen is based on Arcticons and the widget does not really fit into the screen because it's not transparent.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Settings for the widget to set the background color and/or transparency along with the font color (black/white)

oliexdev, 2022-08-20:
sure if somebody send a PR I would merge that.

basings, 2022-09-04:
I was able to change the color to transparent. But I am not yet able to create the functioning color configuration for the widget. If I am ever able to do that, I'll create a PR :)

closed 2025-09-13 by oliexdev