oliexdev/openScale #876: RENPHO ES-CS20M Issues with scales with no device name reported while scanning

AlbertoHSande, 2022-08-23:
Related to [#769](https://github.com/oliexdev/openScale/issues/769)

OpenScale can't list certain models of this scale since it search devices that reports name, but in my case and others as far as I can read, it reports none.

I tried to build a custom apk for debugging and force to connect with a fake device name and a mac address with the QN Scale driver and Trisa Body Analyze 4.0 one, but it gave me always a connection error.

2022-08-23 19:38:07.312 Debug [2] MainActivity: All Bluetooth permissions granted
2022-08-23 19:38:07.334 Debug [2] OpenScale: Trying to connect to bluetooth device [ED:67:39:BF:80:EA] (Renpho)
2022-08-23 19:38:07.338 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2022-08-23 19:38:07.346 Info [2] BluetoothCentralManager: scan started
2022-08-23 19:38:07.347 Debug [2] BluetoothCommunication: Stop machine state
2022-08-23 19:38:08.735 Debug [2] BluetoothCommunication: Found peripheral ''
2022-08-23 19:38:08.739 Info [2] BluetoothCentralManager: scan stopped
2022-08-23 19:38:09.741 Debug [2] BluetoothCommunication: Try to connect to BLE device ED:67:39:BF:80:EA
2022-08-23 19:38:09.851 Info [2] BluetoothPeripheral: connect to '' (ED:67:39:BF:80:EA) using transport LE
2022-08-23 19:38:09.860 Info [2] BluetoothPeripheral: peripheral 'ED:67:39:BF:80:EA' is connecting
2022-08-23 19:38:14.889 Info [116] BluetoothPeripheral: connection failed with status 'CONNECTION_FAILED_ESTABLISHMENT'
2022-08-23 19:38:14.903 Info [116] BluetoothCentralManager: connection to '' (ED:67:39:BF:80:EA) failed
2022-08-23 19:38:14.908 Error [2] BluetoothCommunication: connection '' failed with status 62
2022-08-23 19:38:14.920 Debug [2] MainActivity: Bluetooth connection lost

Will report any progress.

GraciousGpal, 2022-09-17:
See #769 I included code on how to get weight.

oliexdev, 2022-12-08:
closed it here.

closed 2022-12-08 by oliexdev