oliexdev/openScale #877: Crash when trying to connect to scale

Anton-V-K, 2022-08-26:
**Describe the bug**
The crash happens during the initial setup when trying to connect to scale

**To Reproduce**
Steps to reproduce the behavior:
1. Install on a new device (or remove the app's data).
2. Setup users.
3. Connect to scale => Crash!

Reproduced with [version 2.4.6 from F-Droid](https://f-droid.org/ru/packages/com.health.openscale/)

**Expected behavior**
The app should request required permissions and show the list of found scales.

**Additional context**
Realme 8i (Android 11)
Bluetooth is enabled

**Error details**
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-08-26 22:26:47 
Device: Realme RMX3151 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.RuntimeException: Unable to resume activity {com.health.openscale/com.health.openscale.gui.slides.SlideToNavigationAdapter}: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4845)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4883)
	at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:52)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2381)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:261)
	at android.app.ActivityThread.main(ActivityThread.java:8314)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:632)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
Caused by: java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:878)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3080)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.FragmentStore.moveToExpectedState(FragmentStore.java:113)
	at androidx.fragment.app.FragmentManager.moveToState(FragmentManager.java:1374)
	at androidx.fragment.app.FragmentManager.dispatchStateChange(FragmentManager.java:2841)
	at androidx.fragment.app.FragmentManager.dispatchResume(FragmentManager.java:2798)
	at androidx.fragment.app.FragmentController.dispatchResume(FragmentController.java:284)
	at androidx.fragment.app.FragmentActivity.onResumeFragments(FragmentActivity.java:434)
	at androidx.fragment.app.FragmentActivity.onPostResume(FragmentActivity.java:423)
	at androidx.appcompat.app.AppCompatActivity.onPostResume(AppCompatActivity.java:240)
	at android.app.Activity.performResume(Activity.java:8319)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4835)
	... 11 more
```

oliexdev, 2022-12-08:
please enable the openScale debug log and post it here. Unfortunately, I cannot reproduce this error on my smartphone.

closed 2023-01-20 by oliexdev

Anton-V-K, 2023-01-20:
It's strange you cannot reproduce this bug.
I've recently installed same version 2.4.6 on another device with Android 10 (Cubot C20), and the same/similar crash happened in the same situation - when trying to connect to scale without GPS enabled.