oliexdev/openScale #878: FR - manually recalc diagrams if required

aisbergde, 2022-09-01:
**Is your feature request related to a problem? Please describe.**
Today I get wrong statistics and I found the reason: other users used the scale and values have been mixed.
I added a new user, assigned the wrong value to the other user and now the table is fine.
But the diagrams are still wrong, they have not changed after assigning entries to another user.

**Describe the solution you'd like**
There should be a way to manually force the diagrams to recalc.
An intuitive way would be to swipe a finger from top down. Or a point in the menu. Any way to manually force to recalc.

**Describe alternatives you've considered**
A could wait until the next measurement and hope that the diagrams will be recalced.

**Additional context**
The "overview" is correct, it is calculated on the fly.
[image]
[image]

oliexdev, 2022-12-08:
Unfortunately, I cannot reproduce this behavior. Could you provide the necessary steps?

aisbergde, 2022-12-08:
It is hard to reproduce, because it would be required to have the same situation. And on the next day the diagrams have been shown correctly.
That's why it was just a feature request. But it has low priority.