oliexdev/openScale #882: multi user issue with body fat

aisbergde, 2022-09-20:
**Describe the bug**
The Mi scale is used by different persons with different weigh. Normally when importing the measures they are correctly separared by user. And the imported values should not depend on the user selected on the overview tab.
But depending on the selected user at the moment when the app connects to the scale the imported values for body fat are different. They are wrong, if not the person is selected, which currently uses the scale.
Here you see wrong values for the last 2 days. The values are wrong for both users, using the scale.

[image]
[image]

It looks like the issue is related only to body fat. The settings are to import the body fat, but not to calculate.

**To Reproduce**
Steps to reproduce the behavior:
* Use scale by 2 persons with different weight
* enable the settings
** smart user assignment
** ignore out of range data
** merge with last measurement
* select on the overview the person, which will use the scale
* press the Bluetooth icon and step on the scale
** the right values should be imported
* next day (or just later, to avoid merging values)
** in the overview windows select not the person, which will use the scale, but the second person.
**  press the Bluetooth icon and step on the scale
** wrong values for body fat be imported

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build):
latest version, available on f-droid is used
v2.4.6 (61)

**Expected behavior**

* values should be imported consistently not depending on the person, currently selected in the overview window
* correct values should be imported

aisbergde, 2022-09-20:
same issue for visceral fat
[image]

oliexdev, 2022-12-08:
Could you try to disable the "merge with last measurement" option to see if this problem still happens?

aisbergde, 2022-12-08:
I will try tomorrow

aisbergde, 2022-12-09:
I disabled. But is does not read the current value, when I enable Bluetooth and step on the scale. But if I now disable Bluetooth and enable, it imports the last value. I need to experiment a bit in the next days what will be the best workflow.

closed 2023-02-12 by oliexdev