oliexdev/openScale #883: Can't connect on BF800

Dariodom, 2022-09-22:
Hi! I just bought this app since my BF800 can't, as many say, connect to my phone via bluetooth using Beurer Health Manager app.

Unfortunately i cannot connect also using this App, can anyone help me? (I actually don't know if is a bug, a problem or the Scale is not supported)

Ps. I really apprecciate this initiative, i think you are great!

oliexdev, 2022-12-08:
sorry your scale is currently not supported, see https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale

closed 2022-12-08 by oliexdev

Dariodom, 2022-12-08:
But on the link you sent me it says that Beurer BF800 is fully supported!

https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale

oliexdev, 2022-12-09:
sorry you are right but if it doesn't connect with the original vendor app then your scale or your bluetooth is faulty. Did you enable Bluetooth, do you enable location permission? Could you paste the openScale debug log file here when you try to connect to your scale?

Dariodom, 2022-12-09:
I tried everything and nothing seems to work but many people have the same
problem, i think the problem it's not in the scale but in the app itself.