oliexdev/openScale #885: My Scale2, invalid percentage with muscle and some bad view

laltroweb, 2022-09-25:
Like from screen: numbers are not full readable and muscle percentage is crazy 😅
[image]

misibacsi, 2022-10-26:
[image]

I have similar comment, but my problem is with the false colors in the muscle line (see picture).

oliexdev, 2022-12-08:
Do you have multiply measurements on the same day? For the muscle percent please try out the latest dev version

oliexdev, 2022-12-08:
Yes, please do a backup of your data before you install the dev version (use CSV export and do in the settings Backup -> Export Backup)

misibacsi, 2022-12-11:
Hello

I already uninstalled the current software and installed the "dev" version.
(Older data are lost, but this is not important now).

Although the sw version remained the same: 2.4.6 (61)

The problem with the "muscle" is not the percent, but the colored scale.

The colors have a meaning, if I am not mistaken, blue: too low, green:
normal, red: too much.

In case of muscle the software shows green between 20-24 kg, but in reality
it is "too low". The "too low" upper border is near 50 kg. 50+ kg is the
"normal" value.
But in the software the colored line ends near 50 kg (and does not show
bigger numbers).

I send you 2 screenshots: 1 your sw, the other is Mi Body 2 image.
I guess the Mi scale shows the correct values.

Pay attention to values of "muscle" and the scale.

I hope you understand the problem and will correct it.

Regards

Focshole, 2022-12-19:
About the bad muscle data, i think it is a duplicate of #787

closed 2023-01-20 by oliexdev