oliexdev/openScale #887: Xiaomi Mi Body Composition Scale 2

chibbi, 2022-10-11:
**Describe the bug**
I just bought the Xiaomi Mi Body Composition Scale 2, which i intended to use with openScale. Now i opened openScale for the first time (downloaded it a few days ago), to pair the scale with openScale. But when i try to open the Bluethooth tab i get the error, to be seen in Debug log. The error seems to say "i want your location", but that doesn't make any sense in my mind, cause why on earth should an app ask for my location when using bluethooth.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/travis-dev-build): latest f-droid build

**Debug log**
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-10-11 21:57:10 
Device: Xiaomi Redmi Note 7 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.SecurityException: app does not have ACCESS_FINE_LOCATION permission, cannot start scan
	at com.welie.blessed.BluetoothCentralManager.permissionsGranted(BluetoothCentralManager.java:878)
	at com.welie.blessed.BluetoothCentralManager.bleNotReady(BluetoothCentralManager.java:838)
	at com.welie.blessed.BluetoothCentralManager.startScan(BluetoothCentralManager.java:405)
	at com.welie.blessed.BluetoothCentralManager.scanForPeripherals(BluetoothCentralManager.java:515)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.startBluetoothDiscovery(BluetoothSettingsFragment.java:142)
	at com.health.openscale.gui.preferences.BluetoothSettingsFragment.onResume(BluetoothSettingsFragment.java:103)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3071)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:605)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:285)
	at androidx.fragment.app.SpecialEffectsController$FragmentStateManagerOperation.complete(SpecialEffectsController.java:771)
	at androidx.fragment.app.SpecialEffectsController$Operation.completeSpecialEffect(SpecialEffectsController.java:690)
	at androidx.fragment.app.DefaultSpecialEffectsController$SpecialEffectsInfo.completeSpecialEffect(DefaultSpecialEffectsController.java:855)
	at androidx.fragment.app.DefaultSpecialEffectsController.startAnimations(DefaultSpecialEffectsController.java:154)
	at androidx.fragment.app.DefaultSpecialEffectsController.executeOperations(DefaultSpecialEffectsController.java:123)
	at androidx.fragment.app.SpecialEffectsController.executePendingOperations(SpecialEffectsController.java:306)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:1853)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:1764)
	at androidx.fragment.app.FragmentManager.execPendingActions(FragmentManager.java:1701)
	at androidx.fragment.app.FragmentManager$4.run(FragmentManager.java:488)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7590)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
```

chibbi, 2022-10-12:
Welp, found the information in FaQ....

closed 2022-10-12 by chibbi

ohsotech, 2023-03-06:
Mi Smart Scale 2 requires location to be enabled on OnePlus NordCE 5G. This was not the case a few weeks ago.

If I press the bluetooth icon (or have it enabled to auto turn on) it shows this screen, opens the 'enable location' screen and doesn't turn on bluetooth unless location is already on.

[image]

This should only be required for *finding* a bluetooth scale for the first time, no? If I click on my already-paired scale in the settings, it opens the same window, with the small difference that it has a 'no' button too.

So seems like a bug?

FAQ info:

> **Why do openScale needs permission to the coarse location?**
> The answer is simple because the Android API >= 6.0 needs access to the coarse location to search for Bluetooth devices. The App openScale doesn't read your location at any time. Read the official [Android Bluetooth Developer Guide](https://developer.android.com/guide/topics/connectivity/bluetooth-le.html#permissions) or read the [Android 6.0 Changelog](https://developer.android.com/about/versions/marshmallow/android-6.0-changes.html#behavior-hardware-id) for a more technical description.

Doesn't really make sense to me. Especially because in android app settings, location is not allowed for openscale. And if I enable location permission, the exact same thing happens.