oliexdev/openScale #888: is it possinle to sync only one user from the scale?

AlBundy33, 2022-10-22:
II play with the idea to buy a Xiaomi body composition scale 2 (because of the price) or a Beurer BF720 (because it seems that it can be used standalone without the vendors app).

If I create two users on the scale is it possible to sync my user only to openecale on my device and the secind userin another device (e.g. girlfriends phone)?

how does openscale knows what data belongs to what user?
afail BF720 stores the name of the user but only up to 30 datapoints.
So it would be good to know that openscale is capable to select only the data of the correct user because I'm not sure if BF720 deletes the data after sync.

AlBundy33, 2022-11-18:
Currently I have one user in my Beurer BF720 and it works as expected.

I'll ask again if I add an additional user 😄

closed 2022-11-18 by AlBundy33

6b6279, 2024-07-01:
Hello, sorry for bumping this older thread, but I just setup the BF720 with the openScale app and I have noticed that the user I have created on the app wasn't correctly synced with the scale, so that, e.g., the gender of the user on the scale was incorrect. Have you had a similar issue with the BF720?

AlBundy33, 2024-07-20:
I've also created the user on the scale.
OpenScale shows my data correct but I can't remember if I maybe changed the data manually.