oliexdev/openScale #891: Integrate with Android Health Connect

bahuma20, 2022-10-26:
**Describe the solution you'd like**
In an effort to join health data from different apps, Android introduced the Health Connect standard which is a privacy focused local data storage where different fitness apps can read from and write to.

See the documentation here: https://developer.android.com/guide/health-and-fitness/health-connect

This will remove the need for the OpenScale Sync app and will allow integration with much more apps then now are currently supported. For example Google Fit, Fitbit and Samsung Health can read weight data from Health Connect instead of Google Fit being the only integration in OpenScale Sync now.

Because of the local storage and granular permissions available it fits the "Open" ecosystem of the app.

**Describe alternatives you've considered**
Alternatives would be to integrate more services into the OpenScale Sync app which would involve a lot of effort for every service to be integrated and would also require a Cloud connection.

**Additional context**
Currently Samsung Health, Google Fit and just today Fitbit already implement Health Connect. I personally think this is the future of syncing data between Health apps and it will be an industry standard soon.

mauricecyril, 2023-03-22:
This would be amazing!

paolo-caroni, 2023-05-06:
I would also point that the SDK is apache licensed, it's very configurable by the user and privacy friendly. Not dependent on google play service. The data remain on device without an account (except if the user permit the reading by an app that upload data). Other app that use this is myfitnesspal for example.

superhcan, 2023-07-16:
Any news here?

bahuma20, 2023-07-20:
As this issues does not seem to be worked on, i created the app "OpenScale to HealthConnect" which allows you to sync your data from OpenScale to HealthConnect.

It is live in the Play Store since today. So feel free to test it out!

https://play.google.com/store/apps/details?id=io.bahuma.openscale2healthconnect

Any Feedback is welcome. You can find the code for the app here: https://github.com/bahuma20/openscale2healthconnect (Also could be a base for integrating this functionality directly into OpenScale, which i would prefer more)

That repo is also the place to file issues / feedback :)

oliexdev, 2023-07-22:
Nice app @bahuma20 Unfortunately it is written in Kotlin which I don't have any experiences. ;)
Anyway would be great to hear from your Google review process.
Last time I looked into Health Connect, the Google API request form had much more questions. 
I won't prefer to integrate it into openScale directly because it's use Google Play Services which prevents an easily publishing on fdroid. Additionally openScale should no allowed a direct connection to the internet so you health data are safe. But I could integrate someday in openScale sync or users can use of course your app. (Btw which license has your source-code?)

bahuma20, 2023-07-22:
Yes Google Play Services is a problem then.

But the internet connection is not. Health Connect works completely locally and the internet permission is not required to integrate it.

In OpenScale Sync would also be cool. I wanted to make a PR for that app first, but then i noticed that it is not Open Source.

Good that you mention the license for my app. I forgot to add it 😅