oliexdev/openScale #892: Show both % and absolute mass of muscle mass and body fat

bschtl, 2022-10-29:
**Is your feature request related to a problem? Please describe.**
My scale reports body fat and muscle mass as percentage values. I would be quite helpful to mee if openScale would automatically calculate the absolute mass according to bodyweight and also show it in the entry of a measurement, so I don't have to change settings or do it manually.

**Describe the solution you'd like**
In the overview behind the entries of body fat and muscle mass in % it should also show the calculated mass according to the measurement of bodyweight. So if my muscle mass is 40% and the bodyweight 75kg it should show 30kg of muscle mass (perhaps in brackets). The same for body fat and perhaps other values where it makes sense (I am just using these and would like to see it there) and perhaps vice versa, so you input muscle mass in absolute mass and openScale calculates the percentage?

oliexdev, 2022-12-08:
You could do that already in openScale just go into the settings->measurements->(body fast, muscle mass or so on)-> click on the setting symbol next to the measurement (small gear) and disable "Measurement in %"

closed 2022-12-08 by oliexdev

bschtl, 2022-12-14:
Thanks for the reply! I fear you misunderstood me. I know that you can change between percentage and mass. But my question was if it was possible to show **both** values at once, one based on your input (percentage) and one calculated (mass) from it. In my case my scale only measures percentage, but I want to have the absolute mass as a value in openScale, too, shown at the same time. 
It could be displayed in brackets after the percentage value and doens't need to have its own graph, but for me it would be really helpfull if I wouldn't have to change the settings every time for multiple values or calculate the mass from the percentage manually.