oliexdev/openScale #895: Google Fit sign in failed

tdehoog, 2022-11-01:
**Describe the bug**
OpenScale sync loses Google Fit sign in. I need to manually disable and enable Google Fit and do a manual sync to upload weight data to Google Fit.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to openScale sync
2. See "Google Fit sign in failed"
3. Open menu and navigate to Google Fit
4. Disable Google Fit
5. Enable Google Fit
6. Manual sync
7. Wait for unknown period for openScale to lose sign in to Google Fit
8. Go to 1

Reproduced with openScale 0.3.4 (8)

**Expected behavior**
Automatically upload data to Google Fit when a weight is added with openScale Pro.

**Additional context**
Add any other context about the problem here.

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

chrisilutz, 2023-01-23:
I've got the same problem on several devices over years. Device: Pixel 7pro

artemnadein, 2023-03-20:
+1
The same problem, pixel 4a Android 12, pixel 5 Android 13.

superhcan, 2023-07-16:
I have the same problem. OSS looses the Google Fit connection after a short time.

6thSen, 2023-09-27:
Same issue on my Pixel 7.

tiaanv, 2024-01-10:
Same issue here.  This is a paid app.   No response since 2022. DEV?

yyabadaba, 2024-02-09:
Same problem for over a year. After some time ( didn't mange to find out how long exactly (maybe 1hr) it says that it failed and expired and i have to click on sign in and then it signs in. 
Also the status is not relevant because green check mark stays on even in airplane mode.

It says 

2024-02-09 21:50:27.770 Debug [2] a: Check GoogleFit sync status
2024-02-09 21:50:27.773 Debug [2] a: Google Fit successful sign in

Even if i am in airplane mode.

Nothing else in debug log just just this or something like this when i manually sign in:

2024-02-09 17:45:38.509 Debug [2] c: Debug log enabled, openScale sync v0.3.5 (9), SDK 31, Xiaomi M2007J20CG
2024-02-09 17:45:41.154 Debug [2] a: Check GoogleFit sync status
2024-02-09 17:45:41.157 Error [2] a: Google sign in expired
2024-02-09 17:45:41.405 Error [2] a: Google Fit sign in failed
2024-02-09 17:45:42.333 Debug [2] a$b: Google sign in button clicked
2024-02-09 17:45:43.455 Debug [2] a: Google Fit permission granted
2024-02-09 17:45:45.987 Debug [2] c: openScale version 64 with content provider API version 1
2024-02-09 17:45:45.989 Debug [2] a: openScale version is ok(com.health.openscale.pro)
2024-02-09 17:45:45.999 Debug [2] a: openScale user found(xxxxx)
2024-02-09 17:45:46.000 Debug [2] a: Check GoogleFit sync status
2024-02-09 17:45:46.001 Debug [2] a: Google Fit successful sign in

oliexdev, 2025-05-10:
Regarding synchronization with health services, please note that **Google Fit is no longer being actively supported by Google**, and the Google Fit API has been deprecated.

As a result, I have transitioned the synchronization functionality in openScale sync to use **Health Connect** instead. Health Connect is the new standard for managing health and fitness data on Android.

The upcoming beta version (0.4) of openScale sync includes this transition to Health Connect and features a complete rewrite of the application. If you'd like to help test the new Health Connect integration and see if this issue is resolved in the updated version, please consider joining the beta program. You can find more details and instructions on how to join in our pinned beta testing invitation issue here: https://github.com/oliexdev/openScale/issues/1124

closed 2025-05-10 by oliexdev