oliexdev/openScale #898: Unexpected error

vitorgomateus, 2022-11-13:
Trying to set up the app with my weighing balance for the first time and this error shows up. My balance is a Vont Ted Smart Scale VNT-SBS01.

```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-11-13 09:51:28 
Device: Samsung SM-M526BR 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:245)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.app.ActivityThread.main(ActivityThread.java:8751)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
```

oliexdev, 2022-12-08:
sorry your scale is not supported see https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale for supported scales.

closed 2022-12-08 by oliexdev

vitorgomateus, 2022-12-08:
Oh, the error message was not clear. Thank you!