oliexdev/openScale #902: Pixel 7 - MI scale 2 : device not found

esseti, 2022-11-25:
With my new Pixel 7 Pro when I try to connect the MI scale 2 it does not find it out.
It works with my old OnePlus 6t.

Same results using the Zepp app: is it a bluetooth problem of pixel?

vardelean, 2022-11-27:
I'm having a similar issue on my Pixel 7. 
I did a bit of research and it looks like there is an issue with Android 13, see [this](https://issuetracker.google.com/issues/240485116) and [this](https://www.reddit.com/r/android_beta/comments/we2sq1/broadcast_name_for_some_ble_devices_are_null_on/) discussions.

Not sure how to fix this though, some people are suggesting to downgrade the phone's OS to Android 12....

I guess we are going to have to keep reporting this bug to Google until they fix it in the next release.

vardelean, 2022-12-01:
I found a work around for the no-device-name OS13 bug. It works for me, but it may not be for everyone.
Here's my work around:

In the advertising packet of a Bluetooth module (or a modules in a scale), there is a fix number of bytes allocated to the manufacturer data. In my case (I am using a DA14530 chip), 25 bytes were allocated to the manufacturer's device info. 
If any of these bytes is a 0x0, the OS13 interprets that as a null - end of string - and it never proceeds to parsing the next field, which is the scan response and the device name. That's why you cannot see the device name in the scan. 

My solution was to replace all the 0x0 in the manufacturer's name string with something different, like a space (0x20). All of a sudden, the device name showed up and things got back to normal. 

Of course, I was lucky enough to be able to access the Bluetooth module and change the manufacturer name string, and I do realize you can't do that on just any scale, so this work around is not extremally helpful, it just shows that the OS13 scan parsing is not working right and perhaps it will give you some other ideas of how to deal with the bug. 

If you find another work around, please share it everyone, I would be curious to see how other people are dealing with this.

closed 2022-12-08 by oliexdev