oliexdev/openScale #904: Medisana BS444 scale not work correctly

brunop69, 2022-12-08:
Hi

on file BluetoothMedisanaBS44x.java

the app not retrieve the weight, but crash when arrive on the step 4:

[image]
protected boolean onNextStep(int stepNr) {
        switch (stepNr) {
            case 0:
                // set indication on for feature characteristic
                setIndicationOn(WEIGHT_MEASUREMENT_SERVICE, FEATURE_MEASUREMENT_CHARACTERISTIC);
                break;
            case 1:
                // set indication on for weight measurement
                setIndicationOn(WEIGHT_MEASUREMENT_SERVICE, WEIGHT_MEASUREMENT_CHARACTERISTIC);
                break;
            case 2:
                // set indication on for custom5 measurement
                setIndicationOn(WEIGHT_MEASUREMENT_SERVICE, CUSTOM5_MEASUREMENT_CHARACTERISTIC);
                break;
            case 3:
                // send magic number to receive weight data
                long timestamp = new Date().getTime() / 1000;
                if(applyOffset){
                    timestamp -= SCALE_UNIX_TIMESTAMP_OFFSET;
                }
                byte[] date = Converters.toInt32Le(timestamp);

                byte[] magicBytes = new byte[] {(byte)0x02, date[0], date[1], date[2], date[3]};

                writeBytes(WEIGHT_MEASUREMENT_SERVICE, CMD_MEASUREMENT_CHARACTERISTIC, magicBytes);
                break;
            case 4:
                sendMessage(R.string.info_step_on_scale, 0);
                break;
            default:
                return false;
        }

        return true;
    }

I have checked with BLE Scanner the characteristic and the custom service is not BluetoothGattUuid.fromShortCode(0x78b2);
on the image attach you can see that it's different 00007802.
I tried to change it on the code, but now the openSale say please barefoot on the scale.
Can you help me ?

Thanks in advance
Paolo

oliexdev, 2022-12-08:
Maybe you have a different version of the scale!? What happens after you step on the scale? Please attach the openScale debug log here with your changed BLE short code.

brunop69, 2022-12-08:
Hi 

below the openScale debug log:

2022-12-08 19:32:58.023 28483-28483/com.health.openscale D/ActivityThread: handleBindApplication()++ app=com.health.openscale
2022-12-08 19:32:58.024 28483-28483/com.health.openscale D/CompatibilityChangeReporter: Compat change id reported: 171979766; UID 10426; state: ENABLED
2022-12-08 19:32:58.035 28483-28483/com.health.openscale W/ActivityThread: Application com.health.openscale is waiting for the debugger on port 8100...
2022-12-08 19:32:58.037 28483-28483/com.health.openscale I/System.out: Sending WAIT chunk
2022-12-08 19:32:57.545 28483-28483/? I/ealth.openscal: Late-enabling -Xcheck:jni
2022-12-08 19:32:57.839 28483-28483/? E/USNET: USNET: appName: com.health.openscale
2022-12-08 19:32:57.854 28483-28483/? D/ProcessState: Binder ioctl to enable oneway spam detection failed: Invalid argument
2022-12-08 19:32:57.859 28483-28483/? D/ActivityThread: setConscryptValidator
2022-12-08 19:32:57.860 28483-28483/? D/ActivityThread: setConscryptValidator - put
2022-12-08 19:32:57.979 28483-28483/? V/studio.deploy: Startup agent attached to VM
2022-12-08 19:32:57.980 28483-28483/? V/studio.deploy: No existing instrumentation found. Loading instrumentation from instruments-2ca9b3e0.jar
2022-12-08 19:32:57.985 28483-28483/? W/ealth.openscal: DexFile /data/data/com.health.openscale/code_cache/.studio/instruments-2ca9b3e0.jar is in boot class path but is not in a known location
2022-12-08 19:32:57.993 28483-28483/? V/studio.deploy: Applying transforms with cached classes
2022-12-08 19:32:58.017 28483-28483/? W/ealth.openscal: Redefining intrinsic method java.lang.Thread java.lang.Thread.currentThread(). This may cause the unexpected use of the original definition of java.lang.Thread java.lang.Thread.currentThread()in methods that have already been compiled.
2022-12-08 19:32:58.017 28483-28483/? W/ealth.openscal: Redefining intrinsic method boolean java.lang.Thread.interrupted(). This may cause the unexpected use of the original definition of boolean java.lang.Thread.interrupted()in methods that have already been compiled.
2022-12-08 19:32:58.638 28483-28483/com.health.openscale I/System.out: Debugger has connected
2022-12-08 19:32:58.638 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:58.839 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:59.039 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:59.239 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:59.439 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:59.639 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:32:59.839 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:33:00.040 28483-28483/com.health.openscale I/System.out: waiting for debugger to settle...
2022-12-08 19:33:00.240 28483-28483/com.health.openscale I/System.out: debugger has settled (1321)
2022-12-08 19:33:00.242 28483-28483/com.health.openscale W/ActivityThread: Slow operation: 2218ms so far, now at handleBindApplication: Before HardwareRenderer
2022-12-08 19:33:00.258 28483-28483/com.health.openscale W/ActivityThread: Slow operation: 2235ms so far, now at handleBindApplication: After HardwareRenderer
2022-12-08 19:33:00.294 28483-28483/com.health.openscale D/nativeloader: Configuring classloader-namespace for other apk /data/app/~~26iMCAMsOB34eR9U3h3KVA==/com.health.openscale-20YCWslzjqdaRBxsJ1OZ-Q==/base.apk. target_sdk_version=31, uses_libraries=, library_path=/data/app/~~26iMCAMsOB34eR9U3h3KVA==/com.health.openscale-20YCWslzjqdaRBxsJ1OZ-Q==/lib/arm64, permitted_path=/data:/mnt/expand:/data/user/0/com.health.openscale
2022-12-08 19:33:00.328 28483-28483/com.health.openscale V/GraphicsEnvironment: ANGLE Developer option for 'com.health.openscale' set to: 'default'
2022-12-08 19:33:00.328 28483-28483/com.health.openscale V/GraphicsEnvironment: Neither updatable production driver nor prerelease driver is supported.
2022-12-08 19:33:00.334 28483-28483/com.health.openscale D/LoadedApk: LoadedApk::makeApplication() appContext.mOpPackageName=com.health.openscale appContext.mBasePackageName=com.health.openscale
2022-12-08 19:33:00.334 28483-28483/com.health.openscale D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2022-12-08 19:33:00.355 28483-28483/com.health.openscale D/NetworkSecurityConfig: No Network Security Config specified, using platform default
2022-12-08 19:33:00.361 28483-28483/com.health.openscale I/CustomActivityOnCrash: CustomActivityOnCrash has been installed.
2022-12-08 19:33:00.384 28483-28483/com.health.openscale D/ActivityThread: handleBindApplication() --
2022-12-08 19:33:00.404 28483-28543/com.health.openscale D/OpenGLRenderer: RenderThread::requireGlContext()
2022-12-08 19:33:00.406 28483-28543/com.health.openscale I/AdrenoGLES-0: QUALCOMM build                   : 5eaa426211, I07ee46fc66
    Build Date                       : 10/08/21
    OpenGL ES Shader Compiler Version: EV031.32.02.16
    Local Branch                     : 
    Remote Branch                    : 
    Remote Branch                    : 
    Reconstruct Branch               : 
[... cut ...]
2022-12-08 19:33:40.425 28483-28483/com.health.openscale D/MainActivity: Bluetooth scale message: Posizionati sulla bilancia a piedi scalzi
2022-12-08 19:33:40.532 28483-28483/com.health.openscale D/BluetoothCommunication: SUCCESS: Writing <02 02 2E 92 63> to <00008a81-0000-1000-8000-00805f9b34fb>
2022-12-08 19:33:40.533 28483-28483/com.health.openscale D/BluetoothCommunication: Step Nr 5
2022-12-08 19:33:41.140 28483-28483/com.health.openscale D/BluetoothCommunication: Invoke delayed disconnect in 60s

oliexdev, 2022-12-09:
looks generally good. You could add in onBluetoothNotify method a log statement if the scale sends at least something
`Timber.d("Scale output" + byteToString(value));`

brunop69, 2022-12-09:
Hi, In attach the screenshot android studio debug and log file.
[image]
[openscale-debug.txt](https://github.com/oliexdev/openScale/files/10193269/openscale-debug.txt)

brunop69, 2022-12-09:
Hi, One more question this new bodyscale is supported ?

https://www.amazon.it/gp/product/B07WNM9B57/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

Thanks

oliexdev, 2023-01-20:
sorry I give you the wrong Timber.d code, please try it with:

Timber.d("Scale output value: %s (%s)", byteInHex(value), byteToString(value));

The mi scale v2 should be supported but I can't give you a guarantee because I don't own them. Maybe they have change a little bit in this scale.

oliexdev, 2023-02-27:
I will close this issue until we here some feedback.

closed 2023-02-27 by oliexdev