oliexdev/openScale #905: Application crashes when connecting to scale

nmfm, 2022-12-17:
**Describe the bug**
Application crashes when connecting to scale (QN-Scale)

**To Reproduce**
Steps to reproduce the behavior:
I setup a user but was never able to connect to the scale without a crash.
This trace was generated by:
1. Open settings -> Bluetooth
2. Press "Your Bluetooth Scale"
3. List of bluetooth devices, including my scale, is displayed
4. Select scale device

Reproduced with release 2.4.6  
dev build (https://github.com/oliexdev/openScale/releases/tag/dev-build) fails to install on my device

**Expected behavior**
Connection to the scale should be established without crashing

**Debug log**
```
Build version: 2.4.6 
Build date: 1981-01-01 01:01:02 
Current date: 2022-12-17 16:46:37 
Device: OnePlus ONEPLUS A5010 
OS version: Android 10 (SDK 29) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at java.util.Objects.requireNonNull(Objects.java:228)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1202)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:599)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:601)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:453)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:210)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.app.ActivityThread.main(ActivityThread.java:7697)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:516)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
```
[openScale_2022-12-17_16-44.txt](https://github.com/oliexdev/openScale/files/10252202/openScale_2022-12-17_16-44.txt)

oliexdev, 2023-01-20:
It seems you have a slightly different scale than the implemented one. Could you post the Bluetooth services and characteristic of your scale, see Nr. 2 at https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale#2-find-out-the-bluetooth-services-and-characteristic

nmfm, 2023-01-21:
This is the debug log I got following step 2 with the development version (finally got it to install by uninstalling the non-dev version first):
[openScale_2022-12-17_16-44.txt](https://github.com/oliexdev/openScale/files/10472963/openScale_2022-12-17_16-44.txt)

Using BLE Scanner I get some info about 5 services in the scale, but I don't know how to save it except by screenshots
[BLEScan.zip](https://github.com/oliexdev/openScale/files/10473014/BLEScan.zip)

oliexdev, 2023-01-22:
it seems your scale slightly difference from the other QN-scales because the scale don't have the Bluetooth `0xffe3 ` characteristic. see also #771. Without some experiments it's hard to solve the problem without the scale.

nmfm, 2023-01-26:
Would it help if I tried to go through all the steps in https://github.com/oliexdev/openScale/wiki/How-to-reverse-engineer-a-Bluetooth-4.x-scale ?

oliexdev, 2023-01-28:
it depends do you know a bit of java? You could experiment if you just change the Charachteristic in Step 2 in `com.health.openscale.core.bluetooth.BluetoothQNScale` to `0xffe2` for example.

edsenabr, 2023-10-02:
hi, @oliexdev , 

I'm facing the same issue with an scale that I've been using with OpenScale pro for several years now (even before that was a "pro" version); 

I've just moved from a Samsung S21 to an S23 (but the previous one was already@Android 13) and when I try to connect the scale with the phone for the very first time, I receive that error.