oliexdev/openScale #907: Mi Scale 2 isDateInvalid flag uses the same bit 6 of ctrlByte1 as isCattyUnit

pitus, 2023-01-02:
I was trying to understand the data for **Mi Scale 2** from this code and I noticed that the `isDateInvalid` is testing the **same bit 6** of `ctrlByte1` as `isCattyUnit` so one of these must be wrong.
I'm guessing it's the `isDateInvalid` as I've seen elsewhere that `isCattyUnit` is on bit 6 of `ctrlByte1`.
Not sure, however, where `isDateInvalid` info is located - which bit of which byte to test?

https://github.com/oliexdev/openScale/blob/83c2d941ea8ec349c92fba173398c97a5ee45b70/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothMiScale2.java#L144

x11x, 2023-01-16:
I'm pretty sure on Mi Body Composition 2 scale, that bit is `isCattyUnit`. I have changed the units to catty and that bit (bit 6 of second control byte) is set.
Also, that is consistent with what others have found, as you mentioned.
Not sure yet what or where `isDateInvalid` is.
Would that be before the date has been set for the first time (i.e. after batteries removed)?
Or if you have attempted to set an invalid date on it?
(will investigate some more at some point)

x11x, 2023-01-16:
I see this has been asked twice before #652, #286 and closed.

This is not a problem for me as I don't use catty units. I just came across this issue while trying to learn about and implement parsing data for this scale.

I believe it would be a problem for anyone trying to use catty units with this scale, since `isDateInvalid == isCattyUnit` so the `!isDateInvalid` on line 150 would prevent the measurement from being parsed. Obviously nobody is using catty units or they would have reported this.

I will share what I have found trying to investigate what `isDateInvalid` might mean:

I have not spent much time exploring this but I can't figure out what situations the date might be invalid.
I have tried:
 - replacing the batteries
   - time is reset to 1970-01-01T00:00:00, but no different flags are set on the first 2 control bytes-- they are all the same as they would be when it is set to the current date
 - setting time to all zeros
   - everything works as normal (no unusual values in first 2 bytes), the time even increases, even though this is an invalid date
 - setting to a date before 1970
   - again everything is normal

When I change to catty units, ctrl byte 1 bit 6 is definitely always on and the reading is definitely in cattys (0.5 kg), both on the scale display and in BLE advertisement data.

So for my scale (Body Composition Scale 2), I think it is safe to assume bit 6 of ctrl byte 1 is isCattyUnit. Not sure about other models

The only unknown bits i have seen are (counting bits from LSB as 0):
 - first control byte - bit 1 (always on)
 - first control byte - bit 2
   - I have seen this come on sometimes. It is referred to in the [wiecosystem/Bluetooth document](https://github.com/wiecosystem/Bluetooth/blob/master/doc/devices/huami.health.scale2.md#advertisement) document as "partial data". I have not figured out what this means exactly. Pretty sure this is not isDateInvalid as it comes on briefly during measurements and not when changing the date, resetting the batteries.
 - second control byte - bit 2 (always on)

x11x, 2023-01-16:
It is always possible that some models do set this flag when the date is invalid somehow, so changing this could break things for some users. I would be interested to know more about what causes this "invalid date" condition, and the original reason for this code.